espressif/esp-idf #10172: Can support the Additional Controller Advertising Data(ACAD) feature when using the BLE5.0 extended advertising

xiaolongba, 2022-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Experts,

Recently, I was debugging the BLE v5.0 extended advertising. based on the description of **BLUETOOTH SPECIFICATION Version 5.0 | Vol 6, Part B | 2.3.4.8  ACAD field**, Extended advertising can carry the (ACAD), but I cannot find the related APIs that can configure the data of this ACAD field. so in my understanding, the current esp-idf cannot support this feature yet? or any APIs I missed?

Concerning this issue, could you please give me a hand?

thanks in advance.

xiewenxiang, 2022-11-14:
This field is Additional Controller Advertising Data field, there is no standard HCI command to set this field.
If must to set this field, it maybe should use vendor HCI commands.
This is not supported for now.

xiaolongba, 2022-11-14:
Hello @xiewenxiang Bro,

I agree with you, but the Additional Controller Advertising Data field is not out of Bluetooth spec although there is not corresponding HCI command that can configure it.

so it is possible to implement this feature next release? or any workarounds can be used for this purpose?

thanks a lot in advance.

xiewenxiang, 2022-11-15:
Hi @xiaolongba 
In Core_spec 5.2, this field doesn't really have any specific use, there are more detailed instructions in 5.2

Such as:

`BLUETOOTH CORE SPECIFICATION Version 5.2 | Vol 3, Part C.  Page 1371`

`A device in the Broadcast Isochronous Synchronizability mode shall send the BIGInfo in the ACAD field which is located in the AUX_SYNC_IND PDU of periodic advertisement `

The controller populates this field automatically when it populates BIGinfo, Therefore, my understanding is that this field should not be used to actively configure the host. I am not sure if my understanding is correct....

[image]

xiaolongba, 2022-11-15:
Hello @xiewenxiang Bro,

From Bluetooth v5.0 spec, it doesn't support the BIS info which is used for LE audio, but the ACAD field is still present, so in my opinion, BLE v5.0 shall support this even using the host. Could you help me double-check?

Thanks in advance.

xiewenxiang, 2022-11-15:
@xiaolongba  We'll evaluate it and get back to you.

xiewenxiang, 2022-11-15:
Hi @xiaolongba 
We talked about it, and we thought although no specific usage was given in 5.0, subsequent releases have been given specific usage scenarios, so, this field is just for controller.

We also want to know why you want to set this field, The Host Advertising filed should be have enough length for the user to use.

xiaolongba, 2022-11-15:
Hello @xiewenxiang bro,

Indeed, it is only for the controller, is it support this feature in terms of the controller?

The purpose why I asked this question is that I am an educator who is writing some tutorials based on esp32-c3, the link is below:
https://github.com/xiaolongba/HX_DK_FOR_ESP32-C3/wiki 

At this moment, I was trying to explain the new features of the esp32-c3.

So is it possible to give me a hand?

Thanks a lot in advance.

xiewenxiang, 2022-11-15:
Hi @xiaolongba,

You wrote a great tutorial! I'm sure many people will use your tutorial to learn more detail about ESP32-C3.

For the ACAD field, I think you can refer directly to the spec description. It will be auto configured by controller.

xiaolongba, 2022-11-15:
@xiewenxiang Yes, but I can not find which APIs can trigger this operation. Could you give me some tips?
it seems that spec  has not disclosed too much Info about it,Only introduction for the functionality.

xiewenxiang, 2022-11-16:
Yes, the spec does not provide an interface for setting this field, so I think you can explain in the tutorial that this field cannot be set manually.

xiaolongba, 2022-11-16:
> Yes, the spec does not provide an interface for setting this field, so I think you can explain in the tutorial that this field cannot be set manually.

ok, the last question, the current controller of esp32-c3 can support this feature?

xiewenxiang, 2022-11-16:
The controller of C3 currently has code to set this field, but don't support Host to set it.

xiaolongba, 2022-11-16:
> The controller of C3 currently has code to set this field, but don't support Host to set it.

Understood, you can close this ticket.

Thanks.

Alvin1Zhang, 2022-11-16:
Thanks for sharing, feel free to reopen.