espressif/esp-idf #10176: OTA crash when esp_https_ota_finish call

lhjw9810, 2022-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3-devKitC-1

### Power Supply used.

USB

### What is the expected behavior?

ota upgrade successfully

### What is the actual behavior?

ota upgrade crash when esp_https_ota_finish call

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (2030) advanced_https_ota_example: Starting Advanced OTA example
I (3130) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4871, 499
I (3130) esp_https_ota: Starting OTA...
I (3130) esp_https_ota: Writing to partition subtype 16 at offset 0x290000
I (3140) advanced_https_ota_example: Running firmware version: 1.20
I (6130) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4877, 493
I (8340) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4873, 497
W (11100) wifi:<ba-add>idx:1 (ifx:0, 94:d9:b3:e3:11:b9), tid:3, ssn:0, winSize:64
I (11100) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4873, 497
I (13360) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4874, 496
I (17770) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4878, 492
I (22070) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4878, 492
I (24430) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4874, 496
I (26480) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4877, 493
I (28350) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4873, 497
I (33130) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4877, 493
I (37630) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4879, 491
I (39680) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4876, 494
I (41630) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4876, 494
I (43670) HTTP_CLIENT: Body received in fetch header state, 0x3fcf4876, 494
I (44780) esp_image: segment 0: paddr=00290020 vaddr=3c0e0020 size=3ad58h (240984) map
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42003b53
0x42003b53: panic_handler at /Users/hliang/worspace/esp-idf/components/esp_system/port/panic_handler.c:147 (discriminator 3)
```

### More Information.

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x169c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e84
entry 0x403c9904
I (29) boot: ESP-IDF v5.0-rc1-4-g351ba5e342-dirty 2nd stage bootloader
I (29) boot: compile time 12:15:47
I (29) boot: chip revision: v0.1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 8MB
W (54) boot.esp32s3: PRO CPU has been reset by WDT.
W (60) boot.esp32s3: APP CPU has been reset by WDT.
I (65) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (89) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (97) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (104) boot:  3 factory          factory app      00 00 00010000 00271000
I (112) boot:  4 ota_0            OTA app          00 10 00290000 00271000
I (119) boot:  5 ota_1            OTA app          00 11 00510000 00271000
I (127) boot: End of partition table
I (131) boot: Defaulting to factory image
I (136) boot_comm: chip revision: 1, min. application chip revision: 0
I (143) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=3ad58h (240984) map
I (195) esp_image: segment 1: paddr=0004ad80 vaddr=3fc99c00 size=05298h ( 21144) load
I (199) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=d481ch (870428) map
I (356) esp_image: segment 3: paddr=00124844 vaddr=3fc9ee98 size=01028h (  4136) load
I (358) esp_image: segment 4: paddr=00125874 vaddr=40374000 size=15b20h ( 88864) load
I (382) esp_image: segment 5: paddr=0013b39c vaddr=50000000 size=00014h (    20) load
I (391) boot: Loaded app from partition at offset 0x10000
I (392) boot: Disabling RNG early entropy source...
I (403) cpu_start: Pro cpu up.
I (403) cpu_start: Starting app cpu, entry point is 0x403754ec
0x403754ec: call_start_cpu1 at /Users/hliang/worspace/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (418) cpu_start: Pro cpu start user code
I (418) cpu_start: cpu freq: 160000000 Hz
I (418) cpu_start: Application information:
I (421) cpu_start: Project name:     cycle
I (425) cpu_start: App version:      1.20
I (430) cpu_start: Compile time:     Nov 14 2022 12:15:43
I (436) cpu_start: ELF file SHA256:  bff57ebbbcc6f16d...
I (442) cpu_start: ESP-IDF:          v5.0-rc1-4-g351ba5e342-dirty
I (449) heap_init: Initializing. RAM available for dynamic allocation:
I (456) heap_init: At 3FCA69F8 len 00042D18 (267 KiB): D/IRAM
I (463) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (469) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (476) spi_flash: detected chip: gd
I (480) spi_flash: flash io: dio
I (484) sleep: Configure to isolate all GPIO pins in sleep state
I (491) sleep: Enable automatic switching of GPIO sleep configuration
I (498) esp_core_dump_uart: Init core dump to UART
I (503) coexist: coexist rom version e7ae62f
I (508) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (570) pp: pp rom version: e7ae62f
I (570) net80211: net80211 rom version: e7ae62f
I (580) wifi:wifi driver task: 3fceeb54, prio:23, stack:6144, core=0
I (580) system_api: Base MAC address is not set
I (580) system_api: read default base MAC address from EFUSE

hmalpani, 2022-11-14:
Hello @lhjw9810 
I tried the advanced_https_ota example on release/v5.0 and didn't get any errors. Can you share sdkconfig and also if you have changed any code in the example?

lhjw9810, 2022-11-15:
**here is ota code and sdkconfig:**
```
/* Advanced HTTPS OTA example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_http_client.h"
#include "esp_https_ota.h"
#include "nvs.h"
#include "nvs_flash.h"

static const char *TAG = "advanced_https_ota_example";
extern const uint8_t server_cert_pem_start[] asm("_binary_myqcloud_pem_start");
extern const uint8_t server_cert_pem_end[] asm("_binary_myqcloud_pem_end");

#define OTA_URL_SIZE 256

static esp_err_t validate_image_header(esp_app_desc_t *new_app_info)
{
    if (new_app_info == NULL) {
        return ESP_ERR_INVALID_ARG;
    }

    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_app_desc_t running_app_info;
    if (esp_ota_get_partition_description(running, &running_app_info) == ESP_OK) {
        ESP_LOGI(TAG, "Running firmware version: %s", running_app_info.version);
    }

#ifndef CONFIG_EXAMPLE_SKIP_VERSION_CHECK
    if (memcmp(new_app_info->version, running_app_info.version, sizeof(new_app_info->version)) == 0) {
        ESP_LOGW(TAG, "Current running version is the same as a new. We will not continue the update.");
        return ESP_FAIL;
    }
#endif

#ifdef CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK
    /**
     * Secure version check from firmware image header prevents subsequent download and flash write of
     * entire firmware image. However this is optional because it is also taken care in API
     * esp_https_ota_finish at the end of OTA update procedure.
     */
    const uint32_t hw_sec_version = esp_efuse_read_secure_version();
    if (new_app_info->secure_version < hw_sec_version) {
        ESP_LOGW(TAG, "New firmware security version is less than eFuse programmed, %d < %d", new_app_info->secure_version, hw_sec_version);
        return ESP_FAIL;
    }
#endif

    return ESP_OK;
}

static esp_err_t _http_client_init_cb(esp_http_client_handle_t http_client)
{
    esp_err_t err = ESP_OK;
    /* Uncomment to add custom headers to HTTP request */
    // err = esp_http_client_set_header(http_client, "Custom-Header", "Value");
    return err;
[... 2340 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

lhjw9810, 2022-11-15:
> Hello @hmalpani **here is ota code and sdkconfig:**
> 
> ```
> /* Advanced HTTPS OTA example
> 
>    This example code is in the Public Domain (or CC0 licensed, at your option.)
> 
>    Unless required by applicable law or agreed to in writing, this
>    software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
>    CONDITIONS OF ANY KIND, either express or implied.
> */
> #include <string.h>
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "freertos/event_groups.h"
> #include "esp_system.h"
> #include "esp_event.h"
> #include "esp_log.h"
> #include "esp_ota_ops.h"
> #include "esp_http_client.h"
> #include "esp_https_ota.h"
> #include "nvs.h"
> #include "nvs_flash.h"
> 
> static const char *TAG = "advanced_https_ota_example";
> extern const uint8_t server_cert_pem_start[] asm("_binary_myqcloud_pem_start");
> extern const uint8_t server_cert_pem_end[] asm("_binary_myqcloud_pem_end");
> 
> #define OTA_URL_SIZE 256
> 
> static esp_err_t validate_image_header(esp_app_desc_t *new_app_info)
> {
>     if (new_app_info == NULL) {
>         return ESP_ERR_INVALID_ARG;
>     }
> 
>     const esp_partition_t *running = esp_ota_get_running_partition();
>     esp_app_desc_t running_app_info;
>     if (esp_ota_get_partition_description(running, &running_app_info) == ESP_OK) {
>         ESP_LOGI(TAG, "Running firmware version: %s", running_app_info.version);
>     }
> 
> #ifndef CONFIG_EXAMPLE_SKIP_VERSION_CHECK
>     if (memcmp(new_app_info->version, running_app_info.version, sizeof(new_app_info->version)) == 0) {
>         ESP_LOGW(TAG, "Current running version is the same as a new. We will not continue the update.");
>         return ESP_FAIL;
>     }
> #endif
> 
> #ifdef CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK
>     /**
>      * Secure version check from firmware image header prevents subsequent download and flash write of
>      * entire firmware image. However this is optional because it is also taken care in API
>      * esp_https_ota_finish at the end of OTA update procedure.
>      */
>     const uint32_t hw_sec_version = esp_efuse_read_secure_version();
>     if (new_app_info->secure_version < hw_sec_version) {
>         ESP_LOGW(TAG, "New firmware security version is less than eFuse programmed, %d < %d", new_app_info->secure_version, hw_sec_version);
>         return ESP_FAIL;
>     }
> #endif
> 
>     return ESP_OK;
> }
> 
> static esp_err_t _http_client_init_cb(esp_http_client_handle_t http_client)
> {
>     esp_err_t err = ESP_OK;
>     /* Uncomment to add custom headers to HTTP request */
>     // err = esp_http_client_set_header(http_client, "Custom-Header", "Value");
[... 2340 lines cut ...]
> # CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
> CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
> # CONFIG_PPP_SUPPORT is not set
> CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
> CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
> # CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
> # CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
> # CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
> # CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
> CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
> CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
> CONFIG_ESP32_PTHREAD_STACK_MIN=768
> CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
> # CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
> # CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
> CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
> CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
> CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
> # CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
> # CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
> CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
> CONFIG_SUPPORT_TERMIOS=y
> CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
> # End of deprecated options
> ```

hmalpani, 2022-11-15:
Hello @lhjw9810 ,
I tried the advanced_https_ota example with the sdkconfig you provided and the URL present in your sdkconfig. I was able to do the OTA update successfully. I don't think this is a problem in the advanced_https_ota example.

lhjw9810, 2022-11-15:
@hmalpani  ota will conflict with other function, like pwm, gpio,uart Initialization;  close  Initialization function, ota get successfully

closed 2022-11-15 by lhjw9810