espressif/esp-idf #10179: Dynamic Impl: mbedtls_ssl_fetch_input error=29312

Noushadalik, 2022-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

xResult = AzureIoTHubClient_ProcessLoop( &xAzureIoTHubClient,
                                                     sampleazureiotPROCESS_LOOP_TIMEOUT_MS );
  configASSERT( xResult == eAzureIoTSuccess ); 
 Use this command can Receive any incoming MQTT messages from and manage the MQTT connection to IoT Hub.

### What is the actual behavior?

xResult = AzureIoTHubClient_SendTelemetry( &xAzureIoTHubClient,
                                                           ( uint8_t * )device_message, strlen(device_message)+1,
                                                           NULL, eAzureIoTHubMessageQoS1, NULL );
It is not possible to send multiple messages to the IoT hub using this command

### Steps to reproduce.

**Azure IoT middleware for FreeRTOS**
Using below command for publishing messages to IoT hub 
 `xResult = AzureIoTHubClient_SendTelemetry( &xAzureIoTHubClient,( uint8_t * )device_message, strlen(device_message)+1,NULL, eAzureIoTHubMessageQoS1, NULL );
After publishing every message "`AzureIoTHubClient_ProcessLoop(AzureIoTHubClient_t * pxAzureIoTHubClient,uint32_t ulTimeoutMilliseconds)`
use this command for receiving messages from the IoT hub to client

### Debug Logs.

```plain
I (3864018) MQTT_MESSAGE: {"DeviceId":"adapter2","Timestamp": "2022-10-28T13:31:15Z","EnclosureTemp":31.364,"Fan_Status":0,"RTC_Voltage":0.335,"Battery_Voltage":13.255,"Tamper_State":0,"Tank_level":78.548,"Tank_Volume":35346.426,"PH":6.583,"PH_Temperature":28.710}

I (3864038) TIMESTAMP: current timestamp--> 2022-10-28T13:31:15Z, previous timestamp--> 2022-10-28T13:29:56Z

I (3864048) AZ IOT: sending telemetry ...

I (3864068) AZ IOT: Successfully sent telemetry message
I (3864068) AZ IOT: Telemetry sent 

I (3864698) MQTT: Packet received. ReceivedBytes=2.
I (3864698) MQTT: Ack packet deserialized with result: MQTTSuccess.
I (3864698) MQTT: State record updated. New state=MQTTPublishDone.
I (3864718) AZ IOT: Puback received for packet id: 0x00000038
I (3867108) RTC_INDICATOR: RTC error fixed switching indicator off
I (3867108) AZ IOT: Keeping Connection Idle...

E (3887108) SD_CARD: No sd card detected..!!!!
E (3924588) Dynamic Impl: mbedtls_ssl_fetch_input error=29312
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4017514f  PS      : 0x00060530  A0      : 0x80175bde  A1      : 0x3ffe2090  
A2      : 0x3ffbb108  A3      : 0x00008008  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x80175120  A9      : 0x3ffe2070  
A10     : 0x00000000  A11     : 0x3ffe2150  A12     : 0x00000001  A13     : 0x3ffbb1b0  
A14     : 0x3ffe2050  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace:0x4017514c:0x3ffe20900x40175bdb:0x3ffe20b0 0x401e38f9:0x3ffe20e0 0x400e8ab1:0x3ffe2100 0x40146c3d:0x3ffe2130 0x40145821:0x3ffe2150 0x40145940:0x3ffe2180 0x400e64e5:0x3ffe21a0 0x400e627d:0x3ffe21c0 0x400ddc71:0x3ffe21f0 0x40095309:0x3ffe2210 

ELF file SHA256: 5ce97fefa6cf73b7
```

### More Information.

I tried to solve this problem by disconnecting and connecting client from IoT hub after publishing some messages 
also attaching core dump data when the time of crash the code
[core_dump.txt](https://github.com/espressif/esp-idf/files/10002038/core_dump.txt)

AxelLin, 2022-12-24:
error=29312  (0x7280) means MBEDTLS_ERR_SSL_CONN_EOF
https://github.com/espressif/esp-idf/issues/2907#issuecomment-451311378

ESP-YJM, 2023-02-06:
@Noushadalik Has the problem been solved? If not, could you please provide examples that can be reproduced.

AdityaHPatwardhan, 2023-03-13:
Hi, @Noushadalik I am closing the issue due to a lack of response.
Please feel free to re-open if you still face any problem.
Thanks,
Aditya

Advika07, 2023-07-22:
Hi, im facing a issue with this error.
E (5621) Dynamic Impl: mbedtls_ssl_fetch_input error=29312
have tried restarting and rebooting esp but to no avail

Rainbow-Cai, 2024-03-20:
> error=29312 (0x7280) 表示 MBEDTLS_ERR_SSL_CONN_EOF [#2907 （注释）](https://github.com/espressif/esp-idf/issues/2907#issuecomment-451311378)

Hi  I just verified the setting of system properties.
IoTHubMessage_SetContentTypeSystemProperty(msg_handle,
            "application/JSON")

Whenever "/" character is present in system property I am getting the following error.
The Error log report “E (89095) Dynamic Impl: mbedtls_ssl_fetch_input error=29312”

If I use "application@JSON" or "application-JSON" it's working filen. May I know the reason for rejecting "/" character?

Rainbow-Cai, 2024-03-20:
> error=29312 (0x7280) means MBEDTLS_ERR_SSL_CONN_EOF [#2907 (comment)](https://github.com/espressif/esp-idf/issues/2907#issuecomment-451311378)

Hi @AxelLin  Hi I just verified the setting of system properties.
IoTHubMessage_SetContentTypeSystemProperty(msg_handle,
"application/JSON")

Whenever "/" character is present in system property I am getting the following error.
The Error log report “E (89095) Dynamic Impl: mbedtls_ssl_fetch_input error=29312”

If I use "application@JSON" or "application-JSON" it's working filen. May I know the reason for rejecting "/" character?