espressif/esp-idf #10180: GPIO direction setting issue.

matvictor, 2022-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I'm using the ESP32S3-DevKitC-1 V1.1 board (i'ts my first time using the esp32). I'm trying to set a gpio direction : 
> gpio_set_direction(gpio_num_t GPIO_NUM_19, GPIO_MODE_DEF_OUTPUT);

But when i do that, I get an error from the gpio_types.h file.  : 

> "expected declaration specifiers or '...' before '(' token"

[image]

I don't know if I need to change something or if I did anything wrong. Feel free to roast me :). 

My test code : 

[image]
(I putted gpio_reset_pin(...); in comments beacause it agave me some weird type default to int declaration of gpio... errors)

Using ESP-IDF v1.5.1
VScode 1.73.1
Lastest windows update.

atanisoft, 2022-11-14:
Those methods must be called within a function and you shouldn't include `gpio_num_t` in the call to `gpio_set_direction` as it is not valid syntax.

0xjakob, 2022-11-15:
@matvictor Would you mind asking the question on forum (esp32.com)? This place is for people having trouble with a particular part of ESP-IDF. On the forum, you might get better help with setting up examples and other programming questions.

Furthermore, are you actually using an IDF version v1.5.1? If that's correct, please update to the latest version. The currently supported versions are v4.1 and greater.

matvictor, 2022-11-16:
@0xjakob Hi, no I don't mind asking on the esp32 forum at all. + I meant v1.5.1 for the vscode extention. My IDF version is indeed v4.4.2.

matvictor, 2022-11-16:
@atanisoft Yes thank you a lot ! Rookie mistake.

closed 2022-11-16 by matvictor