espressif/esp-idf #10181: can Esp32c3 use ble-mesh？the document is not clear.

Firesuiry, 2022-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32c3/api-guides/esp-ble-mesh/ble-mesh-index.html#get-started-ble-mesh-check-hardware
I have seen above page but the device it use is still [ESP32-DevKitC](https://www.espressif.com/en/products/hardware/esp32-devkitc/overview) and [ESP-WROVER-KIT](https://www.espressif.com/en/products/hardware/esp-wrover-kit/overview)

and [bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server](https://github.com/espressif/esp-idf/tree/v4.4.3/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server) do not support esp32c3.

hipal31, 2022-11-15:
Yes, it supports esp32c3. You first need to set target in your project. Please use command `idf.py set-target esp32c3` to use esp32c3 in your project.

Firesuiry, 2022-11-18:
@hipal31 Thank you!

closed 2022-11-18 by Firesuiry