espressif/esp-idf #10182: Failed to wakeup from light sleep

luacerbb, 2022-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v4.3.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-DevKitC-V4

### Power Supply used.

External 5V

### What is the expected behavior?

gpio can wakeup chip  from light sleep

### What is the actual behavior?

can't wakeup

### Steps to reproduce.

The device failed to wake-up from the deep sleep if the RTC clock source is set to Internal 8.5MHz oscillator.
1、Enable CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256
2、set timer wakeup,
3、SET GPIO16 low level wakeup system,ps GPIO16 as uart2 RX pin
4、enter light sleep
5、not from light sleep

### More Information.

I find person who occur the question in early idf version, but IDF 4.3.4 version  is the lasted V4.3 version,it don't modify the quesion? please help me solve the quesion, thanks.

luacerbb, 2022-11-17:
Thanks for your reply, because v4.3 has other bugs,so I change v4.3 to v4.4.3,it has solve the question.

closed 2022-11-17 by luacerbb

AxelLin, 2022-11-17:
> Thanks for your reply, because v4.3 has other bugs,so I change v4.3 to v4.4.3,it has solve the question.

What kind of bugs in v4.3? Do you have the issue link for tracking?