espressif/esp-idf #10183: E (767) vfs_fat_sdmmc: slot init failed (0x103).

Sunlu2333, 2022-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF_VER: v4.4.1-405-g6c5fb29c2c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I want to store CSI collected by AP to SD card through SPI.

### What is the actual behavior?

E (767) vfs_fat_sdmmc: slot init failed (0x103).
E (768) sd.h: Failed to initialize the card (ESP_ERR_INVALID_STATE).   If you do not have an SD card attached, please ignore this message.  Make sure SD card lines have pull-up resistors in place.

### Steps to reproduce.

I have refer to demo code at ESP32-WROOM-32D, I try setting for SPI pins.
#define PIN_NUM_MISO    2
#define PIN_NUM_CLK    14
#define PIN_NUM_MOSI    15
#define PIN_NUM_CS     13
Its not worked for me, giving error
E (767) vfs_fat_sdmmc: slot init failed (0x103).
E (768) sd.h: Failed to initialize the card (ESP_ERR_INVALID_STATE).   If you do not have an SD card attached, please ignore this message.  Make sure SD card lines have pull-up resistors in place.

I study SPI Pins for ESP-WROOM-32 that are
(VSPI (MISO, CLK, MOSI, CS)=(19, 18, 23, 5))
(HSPI (MISO, CLK, MOSI, CS)=(12, 14, 13, 15))
Also test using VSPI pins, also not work for me. I have used Kingston micro SD 64 GB.

### Build or installation Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6676
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (28) boot: ESP-IDF v4.4.1-405-g6c5fb29c2c 2nd stage bootloader
I (28) boot: compile time 20:27:28
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
[... 202 lines cut ...]
    va_list args;
    va_start(args, format);

#ifdef CONFIG_SEND_CSI_TO_SERIAL
    vprintf(format, args);
#endif

#ifdef CONFIG_SEND_CSI_TO_SD
    if (f != NULL) {
        vfprintf(f, format, args);
    }
#endif

    va_end(args);
}

void sd_flush() {
#ifdef CONFIG_SEND_CSI_TO_SD
    fflush(f);
    fclose(f);
    f = fopen(filename, "a");
#endif
}

#endif //ESP32_CSI_SD_COMPONENT_H

RathiSonika, 2023-06-28:
@Sunlu2333  thank you for reporting the issue here, however I was not able to replicate it on my setup (ESP32-WROOM-32 with SD card adapter) with default SPI pin settings
`#define PIN_NUM_MISO 2`
`#define PIN_NUM_CLK 14`
`#define PIN_NUM_MOSI 15`
`#define PIN_NUM_CS 13`
This [demo example](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/sd_card/sdspi) worked for me.
Can you please again refer [README.md](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/storage/sd_card/sdspi/README.md) to cross check pull up requirement and pin configurations as from above logs and error message, 

`E (767) vfs_fat_sdmmc: slot init failed (0x103).`
`E (768) sd.h: Failed to initialize the card (ESP_ERR_INVALID_STATE). If you do not have an SD card attached, please ignore this message. Make sure SD card lines have pull-up resistors in place.`

I can see, there must be some issue with pull-up resistors. 
Please refer above README.md and let me know for any further query.

RathiSonika, 2023-07-13:
Thank you for reporting the issue on GitHub. Closing this issue as of now as we have not received any update. If you encounter this issue again or have any additional information to share, please don't hesitate to reopen the issue or create a new one. We will be more than happy to assist you further and work towards a resolution.

closed 2023-07-13 by RathiSonika