espressif/esp-idf #10184: Can't add more than 5 services in the ESP-IDF BLE

pedrobotelho15, 2022-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### ESP-IDF example as base:

I'm using this code as a base, a GATT server:

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server

At first everything was fine, with 5 services, but when I added one more service the kernel returned an error status 128. It doesn't even call the events I set up in the big switch-case, in the enum constants esp_gatts_cb_event_t.

### The error:

I did all the configuration of the 6 services correctly. I know this because if I only enter 5 services, all five will work perfectly no matter which one.

In this case, the last service added (the sixth one, no matter which one) gives me an error:

"(1399) BLE TASK: Reg app failed, app_id 0005, status 128"

### The code:

Below is the function that registers the services:
    
    void vInitBluetooth(void) {
        esp_err_t ret;
    
        // Initialize NVS.
        ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            ESP_ERROR_CHECK(nvs_flash_erase());
            ret = nvs_flash_init();
        }
        ESP_ERROR_CHECK(ret);
    
        ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    
        esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
        ret = esp_bt_controller_init(&bt_cfg);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
            return;
        }
    
        ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
            return;
        }
        ret = esp_bluedroid_init();
        if (ret) {
            ESP_LOGE(GATTS_TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
            return;
        }
        ret = esp_bluedroid_enable();
        if (ret) {
            ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
            return;
        }
    
        ret = esp_ble_gatts_register_callback(gatts_event_handler);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gap_register_callback(gap_event_handler);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_ANGLE_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_P_CONTROL_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_I_CONTROL_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_D_CONTROL_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_MODE_CONTROL_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        ret = esp_ble_gatts_app_register(PROFILE_SENSOR_CONTROL_APP_ID);
        if (ret) {
            ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
            return;
        }
        esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
        if (local_mtu_ret) {
            ESP_LOGE(GATTS_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
        }  
    }

I changed the maximum number of services in menuconfig (the variable CONFIG_BT_GATT_MAX_SR_PROFILES) to 9, but it doesn't seem to have any effect.

Does anyone have any ideas?

chegewara, 2022-11-15:
https://github.com/espressif/esp-idf/issues/1644#issuecomment-369189302

xulongzhuang, 2022-11-16:
Hi, @pedrobotelho15 , Please tell me which IDF version (git describe --tags) and which chip you are using.
I have not reproduced the problem you encountered, my test code is as follows, please check if it is different from your code
https://github.com/xulongzhuang/tmp_examples/blob/master/gatt_server/main/gatts_demo.c

In my test code, the setting of CONFIG_BT_GATT_MAX_SR_PROFILES is valid, when CONFIG_BT_GATT_MAX_SR_PROFILES is set to 5 but I registered 6 APP_ID, the same error as yours will appear (error status 128)， But if CONFIG_BT_GATT_MAX_SR_PROFILES is set to 8 or 9, the program can run normally

esp-cjh, 2024-02-28:
@boltragons Does this problem still exist in the latest IDF version?

Jam-cpu, 2024-08-02:
Hi, I have the same problem, regardless of what I set `CONFIG_BT_GATT_MAX_SR_PROFILES` to be it always fails to register the 3rd profile. I have the same error message `Reg app failed, app_id 2, status 128`.

This used to register successfully until I registered a new service before it, therefore hitting the limit.
I am unsure as to why editing `CONFIG_BT_GATT_MAX_SR_PROFILES` does not work. 
I wonder if there is another define overwriting this sdkconfig value?

I am using ESP-IDF v5.1.4

closed 2024-12-17 by Alvin1Zhang