espressif/esp-idf #10187: unable to read coredump from flash

Noushadalik, 2022-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

I wrote the following portion of code to read the coredump from flash

```
void coreD(void){
    const esp_partition_t *pt = NULL;
    size_t bytestoread = 256;
    size_t bytesread = 0;
    char str_dst[256];
   //const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    //assert(partition != NULL);
    pt = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_COREDUMP, "coredump");
    if (pt != NULL)
    {
        size = pt->size;
        printf("POINTER IS NOT NULL \n");
        while(bytestoread > 0)
        {
            printf("STARTING PARTION READ \n");
            printf("Remaining partition size : %zu \n",size);
            esp_err_t er = esp_partition_read(pt, bytesread ,str_dst, bytestoread);
            strcat(coredump,str_dst);
            printf("PARTITION READ \n");
            printf("%s",str_dst);
           if (er != ESP_OK)
            {
                printf("READ IS NOT OK!\n");
                ESP_LOGE(TAG,"Partition read error : %s", esp_err_to_name(er));
                //return ;
            }
           else if (er == ESP_OK)
            {
                printf("\n READ IS OK!\n");
                bytesread = bytesread +256;
                size = size - 256;
 
                if(size <= 256)
                {
                    bytestoread = size;
                }
            }
        
        }
[... 899 lines cut ...]
.coredump.tasks.data 0x3ffbc268 0x158 RW
.coredump.tasks.data 0x3ffbc820 0x1a0 RW
.coredump.tasks.data 0x3ffbc9c8 0x158 RW
.coredump.tasks.data 0x3ffbcf80 0x1a0 RW
.coredump.tasks.data 0x3ffd50c8 0x158 RW
.coredump.tasks.data 0x3ffd4ea0 0x220 RW
.coredump.tasks.data 0x3ffb7660 0x158 RW
.coredump.tasks.data 0x3ffb7490 0x1c8 RW
.coredump.tasks.data 0x3ffb7cc8 0x158 RW
.coredump.tasks.data 0x3ffb7b20 0x1a0 RW
.coredump.tasks.data 0x3ffd5570 0x158 RW
.coredump.tasks.data 0x3ffd6090 0x20c RW
.coredump.tasks.data 0x3ffb7e24 0x158 RW
.coredump.tasks.data 0x3ffbaec0 0x1b8 RW
.coredump.tasks.data 0x3ffd88a0 0x158 RW
.coredump.tasks.data 0x3ffd86a0 0x1f8 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='3669b803'
I (21946) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...
```
**Any help will really be appreciated**

igrr, 2022-11-21:
It's not clear what the `coredump` variable in your code is (it's not declared in the snippet you have posted), but the following part of the code likely won't work:
```c++
            esp_err_t er = esp_partition_read(pt, bytesread ,str_dst, bytestoread);
            strcat(coredump,str_dst);
            printf("PARTITION READ \n");
            printf("%s",str_dst);
```

Core dumps stored in flash are in binary format, therefore:
- using strcat to join binary buffers won't work — strcat works only with 0-terminated C strings, while binary data may contain `\0` bytes anywhere.
- printf("%s") works with 0-terminated C strings, and won't print anything useful for arbitrary binary data

Besides, if `coredump` variable is supposed to contain the entire core dump binary, you may easily run out of memory with this approach.

0xjakob, 2022-12-22:
@Noushadalik any updates on this issue?