espressif/esp-idf #10189: Component esp_http_server not found (4.4.3)

mnlipp, 2022-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to the documentation, all components from esp_idf/components should be available (they all have a CMakeList.txt).

However, including "esp_http_server.h" fails. And from the compile output I can also see that there are a lot of "-Iesp_idf_path/components/...", but there is none for the http server. How do I get this component "activated"?

igrr, 2022-11-16:
@mnlipp indeed, all components are built by default; however you need to specify which components your component depends on. This is described in the [component requirements](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements) section of the programming guide. Please try adding `PRIV_REQUIRES esp_http_server` to the idf_component_register() call in your component's CMakeLists.txt file.

mnlipp, 2022-11-16:
Thanks for the fast answer. I moved on from writing code in main only to writing my first component. And for "main" everything seems to be available "automatically". A bit irritating when you start working with the environment.

closed 2022-11-16 by mnlipp