espressif/esp-idf #10190: Bash session closes on errors after importing IDF environment

KaeLL, 2022-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1826-gac99c0ad6b

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Bash session to remain open

### What is the actual behavior?

After install.sh and export.sh, the bash session abruptly terminates with code 1 after certain events.

### Steps to reproduce.

Running bash 5.2.9 and python 3.11.0, btw.
Just build any app in a way that would cause a compilation error.  
Running 'printenv $IDF_PATH' also triggers it.
Reproducing #10116 (which causes install.sh to fail) also triggers the issue.

igrr, 2022-11-21:
@KaeLL Could you check what is the output of `set -o | grep errexit` in a new terminal, and after running each of the `./install.sh` and `. ./export.sh`?

(Are you by any chance sourcing install.sh as `. ./install.sh`? That would invoke `set -e` in your current shell and cause the  behavior you describe.)

KaeLL, 2022-11-21:
> check what is the output of `set -o | grep errexit`

You caught me. The `install.sh`/`export.sh` routine is done by a script I wrote myself that had a `set -e` at the beginning for some reason. I'm stupid. Thanks.

closed 2022-11-21 by KaeLL