espressif/esp-idf #10191: About esp_pm_get_configuration()

nopnop2002, 2022-11-17:
### Is your feature request related to a problem?

When using esp_pm_get_configuration(), it is necessary to define as follows.

```
#if CONFIG_IDF_TARGET_ESP32
    esp_pm_config_esp32_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32S2
    esp_pm_config_esp32s2_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32C3
    esp_pm_config_esp32c3_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32S3
    esp_pm_config_esp32s3_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32C2
    esp_pm_config_esp32c2_t pm_config = {
#endif
    ESP_ERROR_CHECK( esp_pm_get_configuration(&pm_config) );
```

In the future, when esp32c5 and esp32c6 are added, these will need to be changed.

### Describe the solution you'd like.

I want you to unify the variables of each model.

### Describe alternatives you've considered.

I would like ets_get_cpu_frequency() to work again.

nopnop2002, 2022-11-17:
I found that ets_get_cpu_frequency() can be used in esp-idf v5.1 as well.

```
#include "rom/ets_sys.h"

main() {
	ets_get_cpu_frequency();
}
```

closed 2022-11-17 by nopnop2002

ESP-Marius, 2023-04-27:
As you said, having a single struct per chip was a bit unecessary. This was unified in 8f415a7f448d0438091f2456c173b5e32a829dd1