espressif/esp-idf #10192: bout ROM Library

nopnop2002, 2022-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used ESP-IDF V5 to compile the code below.
```
#include <stdio.h>
#include <string.h>
#include "rom/miniz.h"
#include "rom/ets_sys.h"
#include "rom/tjpgd.h"
#include "rom/md5_hash.h"

void app_main()
{
    uint8_t buf[64];
    memset(buf, 0, sizeof(buf));
    mz_crc32(MZ_CRC32_INIT, (const mz_uint8 *)(buf), 4);
}
```

For some reason, only miniz is a warning.
```
    7 | #warning "{target}/rom/miniz.h is deprecated, please use (#include "miniz.h") instead"
      |  ^~
```

Reference:
https://github.com/espressif/esp-idf/issues/10177

gm-jiang, 2022-11-17:
@nopnop2002 please use `#include` "miniz.h" instead.

nopnop2002, 2022-11-17:
@gm-jiang 

#include "miniz" is not a warning.

I wonder why only miniz becomes a warning.

gm-jiang, 2022-11-17:
@nopnop2002  Since this is a small breaking change that we move `esp_rom/include/{target}/rom/miniz.h` to `esp_rom/include/miniz.h` . In case the user outside has already used `rom/miniz.h` or `esp32/rom/miniz.h`, we put a waring code in original miniz.h for notice that changes.
```
#warning "{target}/rom/miniz.h is deprecated, please use (#include "miniz.h") instead"
#include "../../miniz.h"
```

nopnop2002, 2022-11-17:
@gm-jiang 

Thank you.

small breaking change!!

closed 2022-11-17 by nopnop2002