espressif/esp-idf #10193: Need help with bluetooth

onurdof, 2022-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Connecting windows pc via Bluetooth

### What is the actual behavior?

Pc dont see the device

### Steps to reproduce.

1.Build and flash the ble-mesh (sensor-server) example
2. Try to connect via bluetooth
3. Pc dont see the device name
...

forx157, 2022-11-18:
It doesn't connect via PC at all. You can use the nrf mesh app on your phone to operate.

Campou, 2022-11-28:
Hi @onurdof 

If you want to use the Linux system to provision a Bluetooth Mesh device, you can take this https://www.bluetooth.com/blog/bluez-on-raspberry-pi/ as a reference.

Thanks.

closed 2022-11-28 by Alvin1Zhang