espressif/esp-idf #10194: Failed to initialize the card(ESP_ERR_INVALID_CRC).Make sure SD card lines have pull-up resistors in place.

Sunlu2333, 2022-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF_VER: v4.4.1-405-g6c5fb29c2c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Connect SD card to ESP32-WROVER-IE board through SPI.

### What is the actual behavior?

I (746) sd.h: Using SPI peripheral
I (749) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (750) sdspi_transaction: cmd=52, R1 response: command not supported
I (798) sdspi_transaction: cmd=5, R1 response: command not supported
E (836) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (837) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (838) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (847) sd.h: Failed to initialize the card(ESP_ERR_INVALID_CRC).Make sure SD card lines have pull-up resistors in place.

### Steps to reproduce.

-

### Build or installation Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6676
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.1-405-g6c5fb29c2c 2nd stage bootloader
I (29) boot: compile time 19:05:29
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=291dch (168412) map
I (174) esp_image: segment 1: paddr=00039204 vaddr=3ffb0000 size=03a14h ( 14868) load
I (180) esp_image: segment 2: paddr=0003cc20 vaddr=40080000 size=033f8h ( 13304) load
I (185) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a28cch (665804) map
I (429) esp_image: segment 4: paddr=000e28f4 vaddr=400833f8 size=12a5ch ( 76380) load
I (460) esp_image: segment 5: paddr=000f5358 vaddr=50000000 size=00010h (    16) load
I (471) boot: Loaded app from partition at offset 0x10000
I (471) boot: Disabling RNG early entropy source...
I (483) cpu_start: Pro cpu up.
I (483) cpu_start: Starting app cpu, entry point is 0x4008137c
0x4008137c: call_start_cpu1 at /home/ssr-zenan/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (499) cpu_start: Pro cpu start user code
I (500) cpu_start: cpu freq: 160000000
I (500) cpu_start: Application information:
I (504) cpu_start: Project name:     active-ap
I (509) cpu_start: App version:      1
I (514) cpu_start: Compile time:     Nov 16 2022 19:05:25
I (520) cpu_start: ELF file SHA256:  f665ceba0ef544c6...
I (526) cpu_start: ESP-IDF:          v4.4.1-405-g6c5fb29c2c
I (532) heap_init: Initializing. RAM available for dynamic allocation:
I (539) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (545) heap_init: At 3FFB8DC0 len 00027240 (156 KiB): DRAM
I (551) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (558) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (564) heap_init: At 40095E54 len 0000A1AC (40 KiB): IRAM
I (572) spi_flash: detected chip: generic
I (575) spi_flash: flash io: dio
W (579) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (596) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

-----------------------
ESP32 CSI Tool Settings
-----------------------
PROJECT_NAME: ACTIVE_AP
CONFIG_ESPTOOLPY_MONITOR_BAUD: 115200
CONFIG_ESP_CONSOLE_UART_BAUDRATE: 115200
IDF_VER: v4.4.1-405-g6c5fb29c2c
-----------------------
WIFI_CHANNEL: 4
ESP_WIFI_SSID: myssid
ESP_WIFI_PASSWORD: mypassword
SHOULD_COLLECT_CSI: 1
SHOULD_COLLECT_ONLY_LLTF: 1
SEND_CSI_TO_SERIAL: 1
SEND_CSI_TO_SD: 1
-----------------------

I (746) sd.h: Using SPI peripheral
I (749) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (750) sdspi_transaction: cmd=52, R1 response: command not supported
I (798) sdspi_transaction: cmd=5, R1 response: command not supported
E (836) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (837) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (838) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (847) sd.h: Failed to initialize the card(ESP_ERR_INVALID_CRC).Make sure SD card lines have pull-up resistors in place.
I (862) wifi:wifi driver task: 3ffc1c80, prio:23, stack:6656, core=0
I (865) system_api: Base MAC address is not set
I (870) system_api: read default base MAC address from EFUSE
I (884) wifi:wifi firmware version: 84ed08c
I (885) wifi:wifi certification version: v7.0
I (885) wifi:config NVS flash: enabled
I (887) wifi:config nano formating: disabled
I (892) wifi:Init data frame dynamic rx buffer num: 32
I (896) wifi:Init management frame dynamic rx buffer num: 32
I (902) wifi:Init management short buffer num: 32
I (906) wifi:Init dynamic tx buffer num: 32
I (910) wifi:Init static rx buffer size: 2212
I (914) wifi:Init static rx buffer num: 10
I (918) wifi:Init dynamic rx buffer num: 32
I (923) wifi_init: rx ba win: 6
I (926) wifi_init: tcpip mbox: 32
I (930) wifi_init: udp mbox: 6
I (933) wifi_init: tcp mbox: 6
I (937) wifi_init: tcp tx win: 5744
I (941) wifi_init: tcp rx win: 5744
I (945) wifi_init: tcp mss: 1440
I (949) wifi_init: WiFi IRAM OP enabled
I (954) wifi_init: WiFi RX IRAM OP enabled
I (965) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1067) wifi:mode : softAP (3c:e9:0e:10:1d:45)
I (1071) wifi:Total power save buffer number: 16
I (1072) wifi:Init max length of beacon: 752/752
I (1073) wifi:Init max length of beacon: 752/752
I (1076) wifi:Set ps type: 0
```

Sunlu2333, 2022-11-18:
Well, I have got the solution.
 First, check the voltage supply needed by the SD adapter(Mine is 5V)
Next, add this code:
`host.max_freq_khz = 10000;`
just like this:
[image]
I don't know why, but it works.

closed 2022-11-18 by Sunlu2333

estevaofv, 2022-11-27:
Same issue here, anybody knows how ton increase  above host.max_freq_khz = 10000; ?

SPiernikowski, 2023-02-06:
I had the same problem and changed SPI clock frequency in menuconfik for 20Mhz(the minimum in menuconfig) but it did not help.... 
Sunlu2333 response helped "host.max_freq_khz = 10000" and it is works fine

Supanut-R, 2025-06-12:
I added host.max_freq_khz = 10000;, but it did not work. Could anyone suggest me?