espressif/esp-idf #10198: Incorrect ws_width setting in I2S Philips compatible TDM mode

zjanosy, 2022-11-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-rc1-2-g7f4bcc3695

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

In I2S Philips compatible mode the WS signal is always 1 BCLK.

### What is the actual behavior?

The code sets it to half the frame length (I2S_TDM_AUTO_WS_WIDTH).

### Steps to reproduce.

No specific code is needed. Fix is provided below.

### More Information.

In i2s_tdm.h:

```
#define I2S_TDM_PHILIP_SLOT_DEFAULT_CONFIG(bits_per_sample, mono_or_stereo, mask) { \
    .data_bit_width = (bits_per_sample), \
    .slot_bit_width = I2S_SLOT_BIT_WIDTH_AUTO, \
    .slot_mode = mono_or_stereo, \
    .slot_mask = (mask), \
    .ws_width = I2S_TDM_AUTO_WS_WIDTH, \
    .ws_pol = false, \
    .bit_shift = true, \
    .left_align = false, \
    .big_endian = false, \
    .bit_order_lsb = false, \
    .skip_mask = false, \
    .total_slot = I2S_TDM_AUTO_SLOT_NUM \
}

```

.ws_width should be

`    .ws_width = 1, \`

L-KAYA, 2022-11-18:
1 bit clock width WS signal is identified as PCM_SHORT_SYNC, the macro `I2S_TDM_PCM_SHORT_SLOT_DEFAULT_CONFIG` can support the `philips` that uses `1 BCLK WS`. (Also we have PDM_LONG_SYNC whose WS is one slot width)
As for why we identify the half frame width WS as `philips`,  it's not only can be regard as TDM philips mode, it can also regards as standard two-slot philips when user only registers two slots. Moreover, for some special cases like two slots with 64-bit data width philips, it can also be supported by TDM philips mode by registering 4 32-bit slot. So we think the the `philips` format in TDM, is somehow an expand standard `philips`.

zjanosy, 2022-11-18:
You are right that in stereo I2S (Philips) mode the frame sync is half the frame width. But it may be different for TDM.
I have implemented a driver for the AKM AK4619VN:
https://www.akm.com/eu/en/products/audio/audio-codec/ak4619vn/
This is a 4in/4out channel codec, supporting various TDM modes as well, like TDM128/256 I2S/MSB justified. In I2S mode (which I believe is what you call "Philips" mode) the frame sync is 1 bit, falling edge, 1 clock before the first bit. The PCM_SHORT mode corresponds to the MSB justified mode, where the frame sync is rising edge, aligned with the first bit.
Anyway, it is fairly easy to configure it if one does not use the macro, but the default seems to be not the industry standard way.

L-KAYA, 2022-11-28:
Well, to be honest, there is no unified standard of naming the TDM timing, (e.g. they are called `I2S`, `LJ`, `DSP_A`, 'DSP_B' in es7210 codec, we have an [example on mast3er](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_codec/i2s_es7210_tdm)).

Secondly, our [Technical Reference Manual](https://www.espressif.com.cn/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) has claimed this TDM philips mode,therefore, aligning the name to TRM shouldn't be too much confusing for users.

Anyway, thanks for your advice!

closed 2023-01-12 by Alvin1Zhang