espressif/esp-idf #10200: Error when flashing ESP32-S3 DevKit

dbhaig, 2022-11-18:
Error occurs when flashing ESP32-S3-DevKitC-1 when using master branch:
...
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.0 - v0.99] (this chip is revision v2.8). Use --force to flash anyway.

dobairoland, 2022-11-18:
This issue will be solved by the new esptool release (very soon).

dobairoland, 2022-11-22:
esptool v4.4 has just been released (https://github.com/espressif/esptool/releases/tag/v4.4). It should contain the fix for this. Please re-run the install script and it should be updated to this version.

closed 2022-11-22 by dobairoland