espressif/esp-idf #10201: ble connect err

zongshuwei, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

the idf is 4.41 and the chip as ble host.
        lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
        W (42572) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x3e
        I (42650) state machin: interram  mini 46247
        lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
        W (42662) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x3e
        lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
        W (42752) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x3e
        lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
        W (42842) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x3e
        W (42843) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
        I (42847) ble host manager: ESP_GATTC_DISCONNECT_EVT, reason = 62

zongshuwei, 2022-11-18:
there is the log.

[ble connect err.txt](https://github.com/espressif/esp-idf/files/10038813/ble.connect.err.txt)

xiewenxiang, 2022-11-18:
What is the probability of this issue? 
0x3e error code is Connection failed to be established.  An occasional failure may be normal, but it should be analyzed if it is repeated in a certain scenario.

yaniron, 2022-12-02:
I have a similar issue with 100% probability.
W (6616) BT_HCI: DiscCmpl evt: hdl=1, rsn=0x3e
W (6796) BT_HCI: DiscCmpl evt: hdl=1, rsn=0x3e
W (6966) BT_HCI: DiscCmpl evt: hdl=1, rsn=0x3e
W (7146) BT_HCI: DiscCmpl evt: hdl=1, rsn=0x3e
W (7146) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x3e

yaniron, 2022-12-02:
more input.
the server works in two modes.
if turned on in pairing mode it will pair successfully.
if in scale mode, it will only communicate with the last previously paired device.
--- scenario 1 ---
I turn on the gatt client with an empty nvs, and the server in pairing mode
pairing succeeds.
--- scenario 2 ---
the server remains paired to the client. (bonded)
I turn the gatt client after bonding and turn the server in scale mode
open failed with the 0x3e *always*
-- scenario 3 ---
I run the bonded server in scales mode
I delete the entry from the nvs.
I run the client
it pairs on the fly, and reads the weight correctly.
--
hope it helps in any way

esp-cjh, 2024-02-28:
Hi, @yaniron Does this problem still exist in the latest IDF version? Could you please provide the capture file of the air packets?

closed 2024-12-17 by Alvin1Zhang