espressif/esp-idf #10202: ESP_IDF Set hostname not working

robocircuits, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using esp-IDF and esp-wroom-32 chip. I created a captive portal for esp to set various parameters of esp. currently I access it through the IP address of esp
[image]

I want to access it using the hostname as same as the home assistant uses like this
[image]

I used this code to add a hostname to esp 

[image]

And logs show hostname is set

[image]

but when I enter prashant.local in browser I see nothing. 

Kindly help me if I am doing anything wrong.

chipweinberger, 2022-11-18:
The adapter hostname is a different lower level protocol for naming devices on your network. Look at your router device page.

For http, you need to use MDNS.

gabsuren, 2022-12-05:
@robocircuits thank you for the question. As mentioned @chipweinberger `tcpip_adapter_set_hostname` sets the hostname of an interface. 
In order to set local network hostname you need to use `mDNS` component with the following API `mdns_hostname_set`
You can find the example below (if you use `ESP-IDF version < 5.0`)
[mDNS example](https://github.com/espressif/esp-idf/blob/release/v4.1/examples/protocols/mdns/main/mdns_example_main.c)

If you use `ESP-IDF version >= 5.0` (mDNS component migrated to [esp-protocols](https://github.com/espressif/esp-protocols/))

[mDNS example](https://github.com/espressif/esp-protocols/blob/master/components/mdns/examples/main/mdns_example_main.c)

Please let us know if it helped you or don't hesitate to ask a questions.

Regards,
Suren

closed 2023-02-09 by Alvin1Zhang