espressif/esp-idf #10203: ESP32-S3 rtc_gpio_force_hold_all function link error

imliubo, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2(1b16ef6cfc2479a08136782f9dc57effefa86f66) and v5.1-dev-1264-g1c84cfde14d

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

No error and build success.

### What is the actual behavior?

Link error. **undefined reference to `rtc_gpio_force_hold_all'**

But according this:
[SOC_RTCIO_HOLD_SUPPORTED](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s3/include/soc/soc_caps.h#L264)
esp32-s3 support RTCIO HOLD.
So in this function [gpio_force_hold_all](https://github.com/espressif/esp-idf/blob/master/components/driver/gpio/gpio.c#L720-L729) will call [rtc_gpio_force_hold_en_all]() function.

So ESP32-S3 is support RTCIO HOLD or not?

### Steps to reproduce.

1. git pull
2. write simple test code
3. build with idf.py

### Build or installation Logs.

```plain
/path/to/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/driver/libdriver.a(gpio.c.obj): in function `gpio_force_hold_all':
/path/to/esp-idf/components/driver/gpio/gpio.c:723: undefined reference to `rtc_gpio_force_hold_all'
```

### More Information.

simple test code:
```code
#include <stdio.h>
#include <driver/rtc_io.h>

void app_main(void)
{
    gpio_force_hold_all();
}

```

songruo, 2022-11-21:
This has been fixed on master few days ago (d0a7dc3e9f67f342048e459d787502d0970eaadc). Will backport to earlier IDF version very soon.

imliubo, 2022-11-22:
OK, thanks!

I will close this issue.

closed 2022-11-22 by imliubo