espressif/esp-idf #10205: ESP32-S3 ROM not provide gpio_input_get_high and gpio_output_set_high function?

imliubo, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2(1b16ef6cfc2479a08136782f9dc57effefa86f66) and v5.1-dev-1264-g1c84cfde14d

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

build success.

### What is the actual behavior?

undefined reference to ‘gpio_output_set_high' and ‘gpio_input_get_high’.

Is this not support or missing?

### Steps to reproduce.

1. pull
2. write simple code
3. build with idf.py
...

### Build or installation Logs.

```plain
/path/to/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(test.c.obj):(.literal.app_main+0x0): undefined reference to `gpio_input_get_high'
/path/to/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(test.c.obj):(.literal.app_main+0x4): undefined reference to `gpio_output_set_high'
```

### More Information.

```c
#include <stdio.h>
#include "esp32s3/rom/gpio.h"

void app_main(void)
{
    gpio_input_get_high();
    gpio_output_set_high(0, 0, 0, 0);
}

```

suda-morris, 2022-11-21:
ROM functions are usually not considered public. It's common that some function exists on ESP32 but disappears in the future chips.

Can we know the reason why use the ROM functions instead of using the GPIO driver API like `gpio_get_level`? Are you expecting a function to read multiple adjacent GPIO values？

imliubo, 2022-11-21:
In fact, in [this project](https://github.com/lvgl/lv_binding_micropython), we use a [script](https://github.com/lvgl/lv_binding_micropython/blob/master/gen/gen_mpy.py) to parse and convert the function definition of gpio.h into micropython method. Since these two functions only have definitions and no actual function bodies, an error will be reported during the linking stage. But esp32 rom has provide this so just wanna make sure this is not support or missing, if not support in esp32-s3 we will solve this error in another way.

Related issue: [here](https://github.com/lvgl/lv_binding_micropython/pull/243#issuecomment-1320958393)

gm-jiang, 2022-11-21:
@imliubo Yes, in fact, the ROM code of the S3 also contains the functions `gpio_input_get_high() and  gpio_output_set_high` but it is not exported through the ROM ld interface, I think we can these two functions as S3 ROM patch,but a little RAM consumption. 
one more things, ESP32/S2/S3 ROM support `gpio_input_get_high() and  gpio_output_set_high` since the gpio num of these chips is greater than 32. ESP32-C2/C3/C6/H2(MP) ROM does not support, maybe  we should remove the definitions from `gpio.h`.

imliubo, 2022-11-22:
Hi @gm-jiang ,
> es, in fact, the ROM code of the S3 also contains the functions gpio_input_get_high() and gpio_output_set_high but it is not exported through the ROM ld interface, I think we can add these two functions as S3 ROM patch,but a little RAM consumption.

So this need update idf or something?Then we use this two function with esp32-s3?

> one more things, ESP32/S2/S3/C6 ROM support gpio_input_get_high() and gpio_output_set_high since the gpio num of these chips is greater than 32. ESP32-C2/C3/H2(MP) ROM does not support, maybe we should remove the definitions from gpio.h.

If ESP32-C2/C3/H2(MP) not support i think remove the definitions from gpio.h is better, but this is your guys decision. For us just want build no error.

closed 2022-12-22 by espressif-bot