espressif/esp-idf #10206: Android 13 changed GATT_MAX_ATTR_LEN to 512

phatpaul, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Android devs recently decided to change the data buffer size [GAT_MAX_ATTR_LEN](https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h;l=250;drc=1918034a0730839c7a07b1260b1ab74b80d6b4e6;bpv=1;bpt=1) to 512 bytes.  (was 600)

I was using this [#define ESP_GATT_MAX_ATTR_LEN 600](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/include/api/esp_gatt_defs.h#L295) in my app to set the size of a BLE data transfer.  But recently my app is broken on Android 13 because 2 bytes are being silently dropped and corrupting my BLE data transfer.  
I suggest to change ESP_GATT_MAX_ATTR_LEN to 512 in this library to match what Android did.

FYI this was helpful in tracking down my problem: https://github.com/dariuszseweryn/RxAndroidBle/pull/808

xiewenxiang, 2022-11-21:
In ESP-IDF, `ESP_GATT_MAX_ATTR_LEN_MAX` refers to the maximum length of the attribute value supported by the ble protocol stack.

In app code, user can specify the maximum length of the Attribute value, such as 50, 100, 150..., just does not exceed the maximum value defined by the macro. 

So how much of this macro is defined should not cause any other problems, if other issues are caused, you can provide the exception log, we need to analyze it concretely

phatpaul, 2022-11-21:
You're right, I can define whatever length of Attr value I want to in app code.  I'm not saying it's a bug in ESP-IDF.  

I guess it is a bug in Android (that I doubt they will fix).  If I define it greater than 512 (and fill the buffer), then it will be incompatible with Android 13.  Android 13 just silently truncates the value to 512 bytes.

I had implemented a BLE gatt service to transfer files over BLE in chunks of 512+2 bytes  (512B data + 2B header)
https://github.com/eagi223/esp-idf_Bluetooth_Multi-Service/blob/master/main/wireless/ble_services/ble_service_ftp.c#L264  (and the companion cordova app also used 512+2 bytes chunks: https://github.com/phatpaul/ble-ftp-cordova-example/blob/main/www/js/ble-ftp.js#L690 )  Now on Android 13, the file transfer is corrupted.  The files are missing 2 bytes at each interval of 512 bytes.

Is there anything that could be done on ESP-IDF side to avoid this pitfall?

xiewenxiang, 2022-11-24:
You can try setting the maximum MTU size that can be supported locally when initializing the Bluetooth stack, such as 500.
But I'm not sure it can be avoided..

```c
esp_err_t esp_ble_gatt_set_local_mtu (uint16_t mtu);
```

phatpaul, 2022-12-06:
Thanks for the help.  Closing.

closed 2022-12-06 by phatpaul