espressif/esp-idf #10207: WiFi example doesn´t work with ESP32-C3 - IP_EVENT_STA_GOT_IP never get called

Kampi, 2022-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.2.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### What is the expected behavior?

Print out the IP address given by a DHCP. But the `IP_EVENT_STA_GOT_IP` event isn´t called, so the connection bit in the WiFi STA example doesn´t work. Is this the correct behaviour?

### What is the actual behavior?

I got this output:

```
I (551) wifi:connected with <SSID>, aid = 1, channel 10, BW20, bssid = e8:48:b8:e1:62:2a
I (551) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (561) wifi:pm start, type: 1

I (561) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (571) WiFi: Event ID: 4
I (571) WiFi: WiFi Event...
I (581) wifi:BcnInt:51200, DTIM:1
```

And nothing happens.

### Steps to reproduce.

1. Step
Compile the code

```
#include <esp_log.h>
#include <esp_wifi.h>
#include <esp_check.h>
#include <esp_netif.h>
#include <wifi_provisioning/manager.h>
#include <wifi_provisioning/scheme_ble.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/event_groups.h>

#include <sdkconfig.h>

#define WIFI_FAIL_BIT                           BIT1
#define WIFI_CONNECTED_BIT                      BIT0

static uint8_t _WiFi_Retries                    = 0;

static esp_netif_t* _WiFi_NetIf;
static wifi_init_config_t _WiFi_Config          = WIFI_INIT_CONFIG_DEFAULT();
static EventGroupHandle_t _WiFi_EventGroup;
static wifi_prov_mgr_config_t _WiFi_Prov_Config = {
    .scheme = wifi_prov_scheme_ble,
    .scheme_event_handler = WIFI_PROV_EVENT_HANDLER_NONE,
    .app_event_handler = WIFI_PROV_EVENT_HANDLER_NONE
};

static const char* TAG 							= "WiFi";

static void _WiFi_on_Event_Handler(void* p_Arg, esp_event_base_t EventBase, int32_t EventID, void* p_EventData)
{
    ESP_LOGI(TAG, "Event ID: %i", EventID);

    if(EventBase == WIFI_PROV_EVENT)
    {
        ESP_LOGI(TAG, "Prov Event...");

        switch(EventID)
        {
            case WIFI_PROV_START:
            {
                ESP_LOGI(TAG, "Provisioning started");

                break;
            }
            case WIFI_PROV_CRED_RECV:
            {
                wifi_sta_config_t* wifi_sta_cfg = (wifi_sta_config_t*)p_EventData;

                ESP_LOGI(TAG, "Received Wi-Fi credentials\n\tSSID     : %s\n\tPassword : %s", (const char*)wifi_sta_cfg->ssid, (const char*)wifi_sta_cfg->password);

                break;
            }
            case WIFI_PROV_CRED_FAIL:
            {
                wifi_prov_sta_fail_reason_t* reason = (wifi_prov_sta_fail_reason_t*)p_EventData;

                ESP_LOGE(TAG, "Provisioning failed!\n\tReason : %s"
                         "\n\tPlease reset to factory and retry provisioning",
                         (*reason == WIFI_PROV_STA_AUTH_ERROR) ?
                         "Wi-Fi station authentication failed" : "Wi-Fi access-point not found");
                break;
            }
            case WIFI_PROV_CRED_SUCCESS:
            {
                ESP_LOGI(TAG, "Provisioning successful");

                break;
            }
            case WIFI_PROV_END:
            {
                wifi_prov_mgr_deinit();

                break;
            }
        }
    }
    else if(EventBase == WIFI_EVENT)
    {
        ESP_LOGI(TAG, "WiFi Event...");

        switch(EventID)
        {
            case WIFI_EVENT_STA_START:
            {
                ESP_LOGI(TAG, "WiFi start event...");

                esp_wifi_connect();

                break;
            }
            case WIFI_EVENT_STA_DISCONNECTED:
            {
                if(_WiFi_Retries < CONFIG_WIFI_CON_RETRIES)
                {
                    _WiFi_Retries++;
                    ESP_LOGI(TAG, "Retry %u / %u...", _WiFi_Retries, 10);
                    esp_wifi_connect();
                }
                else
                {
                    ESP_LOGE(TAG, "WiFi connected failed!");

                    xEventGroupSetBits(_WiFi_EventGroup, WIFI_FAIL_BIT);
                }

                ESP_LOGE(TAG, "Can not connect with AP!");

                break;
            }
        }
    }
    else if(EventBase == IP_EVENT)
    {
        ESP_LOGI(TAG, "IP Event...");

        switch(EventID)
        {
            case IP_EVENT_STA_GOT_IP:
            {
                _WiFi_Retries = 0;
                ESP_LOGI(TAG, "Got IP:" IPSTR, IP2STR(&((ip_event_got_ip_t*)p_EventData)->ip_info.ip));
                xEventGroupSetBits(_WiFi_EventGroup, WIFI_CONNECTED_BIT);

                break;    
            }
        }
    }
}

static esp_err_t WiFi_OpenConnection(std::string SSID = std::string(), std::string Password = std::string())
{
    wifi_config_t Config;

    memset(&Config, 0, sizeof(wifi_config_t));
    memcpy((char*)Config.sta.ssid, SSID.c_str(), SSID.length());
    memcpy((char*)Config.sta.password, Password.c_str(), Password.length());
    Config.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;

    ESP_RETURN_ON_ERROR(esp_wifi_set_mode(WIFI_MODE_STA), TAG, "");
    ESP_RETURN_ON_ERROR(esp_wifi_set_config(WIFI_IF_STA, &Config), TAG, "");
    ESP_RETURN_ON_ERROR(esp_wifi_start(), TAG, "");

    return ESP_OK;
}

static esp_err_t WiFi_Connect(std::string SSID, std::string Password, bool Reconnect = false)
{
	uint32_t Timeout = 0;

       ...

    	ESP_LOGI(TAG, "Opening a new WiFi connection...");
        ESP_LOGI(TAG, " SSID: %s", SSID.c_str());
        ESP_LOGI(TAG, " Password: %s", Password.c_str());

        if(WiFi_OpenConnection(SSID, Password) != ESP_OK)
        {
            return ESP_FAIL;
        }

	Timeout = esp_timer_get_time() / 1000ULL;
    do
    {
		if(((esp_timer_get_time() / 1000ULL) - Timeout) >= (10 * 1000UL))
		{
            ESP_LOGE(TAG, "Timeout!");

			return ESP_ERR_TIMEOUT;
		}

		vTaskDelay(10 / portTICK_RATE_MS);
	} while(WiFi_isConnected() == false);

    ESP_LOGI(TAG, "Connected");

    return ESP_OK;
}

static esp_err_t WiFi_BasicInit(void)
{
    std::string HostName;
    wifi_init_config_t Config;

    esp_wifi_disconnect();
    esp_wifi_stop();

    if(_WiFi_EventGroup == NULL)
    {
        _WiFi_EventGroup = xEventGroupCreate();
    }

    esp_event_loop_create_default();

    ESP_RETURN_ON_ERROR(esp_event_handler_register(WIFI_PROV_EVENT, ESP_EVENT_ANY_ID, &_WiFi_on_Event_Handler, NULL), TAG, "");
    ESP_RETURN_ON_ERROR(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &_WiFi_on_Event_Handler, NULL), TAG, "");
    ESP_RETURN_ON_ERROR(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &_WiFi_on_Event_Handler, NULL), TAG, "");

    Config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_RETURN_ON_ERROR(esp_wifi_init(&Config), TAG, "");

    HostName = "tl_" + WiFi_GetMAC();
    ESP_LOGI(TAG, "Set host name: %s", HostName.c_str());
    ESP_RETURN_ON_ERROR(esp_netif_set_hostname(_WiFi_NetIf, HostName.c_str()), TAG, "");

    return ESP_OK;
}

esp_err_t WiFi_Init(void)
{
    std::string SSID;
    std::string Password;

    ESP_RETURN_ON_ERROR(esp_netif_init(), TAG, "");
    _WiFi_NetIf = esp_netif_create_default_wifi_sta();

    #if((not defined CONFIG_GENERAL_DEFAULT_WIFI_SSID) or (not defined CONFIG_GENERAL_DEFAULT_WIFI_PASS))
        #error "Setup SSID or Password not defined!"
    #endif

    ESP_ERROR_CHECK(WiFi_BasicInit());

    SSID = CONFIG_GENERAL_DEFAULT_WIFI_SSID;
    Password = CONFIG_GENERAL_DEFAULT_WIFI_PASS;

    return WiFi_Connect(SSID, Password);
}

void WiFi_Deinit(void)
{
    esp_wifi_disconnect();
    esp_wifi_stop();
    esp_wifi_deinit();
    esp_netif_destroy(_WiFi_NetIf);
    esp_netif_deinit();
}

bool WiFi_isConnected(void)
{
    if(_WiFi_EventGroup == NULL)
    {
        return false;
    }

    return (bool)xEventGroupWaitBits(_WiFi_EventGroup, WIFI_CONNECTED_BIT, pdFALSE, pdFALSE, 10 / portTICK_PERIOD_MS);
}
```

felmue, 2022-11-19:
Hello @Kampi 

have you tried the stock WiFi [station](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) example?

Thanks
Felix

Kampi, 2022-11-22:
> Hello @Kampi
> 
> have you tried the stock WiFi [station](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) example?
> 
> Thanks Felix

Hi @felmue 

I can´t reproduce the issue with the WiFi STA example. Now I have to compare both parts :)

closed 2022-11-22 by Kampi