espressif/esp-idf #10209: [WiFiGeneric.cpp:715] espWiFiStart(): esp_wifi_start 257

kraa965, 2022-11-20:
when starting wi-fi in STA mode, it tries to connect, but freezes, tell me, what could be the problem?

Xiehanxin, 2022-11-21:
hello, @kraa965, can you provide more details? like whic which chip, the IDF version and the error log

kraa965, 2022-11-21:
> hello, @kraa965, can you provide more details? like whic which chip, the IDF version and the error log

Chip: esp32dev
IDF: 4.4.2
Error: espWiFiStart(): esp_wifi_start 257

I uploaded the access point data to the wificonfig.json file, the data is read, but when they get into wifi.begin(witiconfig.ssid, wificonfig.password), the espWiFiStart() error appears: esp_wifi_start 257, and a hang occurs

Xiehanxin, 2022-11-21:
I have checked the error code in ESP arduino and the 257, which is 0x101, means out of memory, you can checkout your remain memory when you start the wifi, and if you try the STA example, can you connect to the AP?
[image]

kraa965, 2022-11-21:
> I have checked the error code in ESP arduino and the 257, which is 0x101, means out of memory, you can checkout your remain memory when you start the wifi, and if you try the STA example, can you connect to the AP?
> [image]
> 

I'm running in sta mode, yesterday I found that it was a lack of memory, I tried to work with partitions, but it didn't work out, I need partitions for spiffs and the code itself, but I'm new to this, can you tell me how to correctly calculate the necessary partitions and how to fill them correctly?

kraa965, 2022-11-22:
I was able to achieve a little success, now at least it does not crash when starting wifi, but I still get an error

> E (24441) wifi:Expected to init 4 rx buffer, actual is 3
> [ 23763][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (24449) wifi:Expected to init 4 rx buffer, actual is 0
> [ 23771][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> [ 23772][E][WiFiSTA.cpp:227] begin(): STA enable failed!
> E (29464) wifi:Expected to init 4 rx buffer, actual is 0
> [ 28786][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (29471) wifi:Expected to init 4 rx buffer, actual is 0
> [ 28793][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> [ 28796][E][WiFiAP.cpp:141] softAP(): enable AP first!
--------------------------------------
updated the sections, now it looks like this

> # Name,   Type, SubType, Offset,  Size, Flags  
> app0,     app,  ota_0,   0x10000, 2M,  
> nvs,      data, nvs,     ,  400K,    
> spiffs,   data, spiffs,  ,  1500K,
----------------------------------------
can you tell me what else I need to fix so that wi-fi starts running?

kraa965, 2022-11-23:
hi @Xiehanxin 
the ESP.getMinFreeHeap function shows

> [INFO] BLE Keyboard action received: 11 1 
> Free Heap before wifi launch: 71620
> E (43743) wifi:Expected to init 4 rx buffer, actual is 3
> [ 43067][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (43752) wifi:Expected to init 4 rx buffer, actual is 0
> [ 43075][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
>  10
>  8
>  6
>  4
>  2
> E (46277) wifi:Expected to init 4 rx buffer, actual is 0
> [ 45600][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (46284) wifi:Expected to init 4 rx buffer, actual is 0
> [ 45607][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> [ 45609][E][WiFiAP.cpp:141] softAP(): enable AP first!
> [WARNING]: Could not connect to AP, so started as AP.
-------------------------------------------------------------------
the esp_get_minimum_free_heap_size() function shows

> [INFO] BLE Keyboard action received: 11 1 
> Free Heap before wifi launch: 53528
> E (19188) wifi:Expected to init 4 rx buffer, actual is 3
> [ 18512][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (19197) wifi:Expected to init 4 rx buffer, actual is 0
> [ 18520][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> [ 18521][E][WiFiSTA.cpp:227] begin(): STA enable failed!
> Free Heap after starting wifi: 53528
> startWifiStation(): attempts = 10
> startWifiStation(): attempt delay = 500
>  10
>  8
>  6
>  4
>  2
> E (21722) wifi:Expected to init 4 rx buffer, actual is 0
> [ 21045][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> E (21729) wifi:Expected to init 4 rx buffer, actual is 0
> [ 21052][E][WiFiGeneric.cpp:678] wifiLowLevelInit(): esp_wifi_init 257
> [ 21054][E][WiFiAP.cpp:141] softAP(): enable AP first!
> [WARNING]: Could not connect to AP, so started as AP.

Xiehanxin, 2022-11-29:
hi @kraa965  sorry for late reply, I think it is caused that your BLE running all the time which take too much memory. can you provide your sdkconfig? maybe you divide much resource to the wifi, what's more, you can use esp_get_free_internal_heap_size to get internal free heap which is allocated for wifi

kraa965, 2022-11-29:
@Xiehanxin thanks for the solution with ble

closed 2022-11-29 by kraa965