espressif/esp-idf #10210: ESP32-S3-N8R2 SDMMC in 4-bit does not run on 40 Mhz with IDF 4.4.2

NARRob, 2022-11-21:
> @ThanosSiopoudis @ntwallace Thank you for reporting the issue! Could you please try to apply the attached patch and see if it improves the situation?
> [0001-Draft-sdmmc-fix-unreasonable-i-o-clock-phase-setting.patch.txt](https://github.com/espressif/esp-idf/files/8236572/0001-Draft-sdmmc-fix-unreasonable-i-o-clock-phase-setting.patch.txt)

@igrr Can this patch be applied to Stable 4.4.2?  Facing the same issue with ESP32-S3-N8.   Compiling with latest master is ok, compiling with 4.4.2 can only mount if 20 MHz and always fails on 40 MHz.   Using SDMMC/4-bit.  We must use the 4.4.2 as the ESP-ADF is not yet fully supported with v5.1 latest (any time frame for v5 to support ADF? or patches?)

Thanks

_Originally posted by @NARRob in https://github.com/espressif/esp-idf/issues/8521#issuecomment-1225803453_

NARRob, 2022-11-21:
Is there a patch for ESP32-S3 for SDMMC to run at 40 Mhz.   The same exact code runs 40 MHz when compiled with v5.1, but fails to mount when compiled with 4.4.2 or 4.4.3.  It only can run on 20 MHz.
We need to use 4.4.2 or 4.4.3 for ESP-ADF code.

chegewara, 2022-11-21:
I am building and running with arduino (esp-idf v4.4.3) sd_mmc using native esp-idf API and it is working, both, SPI and mmc 4-bits. SPI reporting max 20MHz and mmc 40MHz with my sd card.
```
        esp_vfs_fat_sdmmc_mount_config_t mount_config = {
            .format_if_mount_failed = true,
            .max_files = 5,
            .allocation_unit_size = 16 * 1024
        };
        _is_mmc = mmc;

        if(mmc)
        {
            sdmmc_host_t host = SDMMC_HOST_DEFAULT();
            host.max_freq_khz = SDMMC_FREQ_52M;
            host.flags &= ~SDMMC_HOST_FLAG_DDR;
            ret = esp_vfs_fat_sdmmc_mount(path, &host, &slot_config_mmc, &mount_config, &_card);
        }
        else
        {
            sdmmc_host_t host = SDSPI_HOST_DEFAULT();
            host.max_freq_khz = SDMMC_FREQ_DEFAULT;
            spi_bus_initialize((spi_host_device_t)host.slot, &bus_cfg, SDSPI_DEFAULT_DMA);
            slot_config_spi.host_id = (spi_host_device_t)host.slot;
            ret = esp_vfs_fat_sdspi_mount(path, &host, &slot_config_spi, &mount_config, &_card);
        }
```

NARRob, 2022-11-21:
@chegewara Thanks.  You message did not indicate if it was the ESP32-S3.   Maybe adding the ESP-ADF got something to do with it.  I will try using 4.4.3 alone without ADF to see if that's the case.  Thanks

chegewara, 2022-11-21:
Yes, of course it is S3. Im just not sure right now if its with or without PSRAM, because i have few different devkits, including N32R8 OPI mode. Here you can see pins i am using, but S3 can use any free pins with matrix:
```
#define SD_CMD	1
#define SD_CLK  2
#define SD_DAT0 4
#define SD_DAT1 5
#define SD_DAT2 6
#define SD_DAT3 7
```

igrr, 2023-04-17:
The fix has been merged internally and will be available on Github soon. For reference, commit ID will be https://gitlab.espressif.cn:6688/espressif/esp-idf/-/commit/ef64e4e5b3fb4878ed7df62a054f44ae5b131519.

closed 2023-04-17 by igrr