espressif/esp-idf #10212: How to disable esp-idf 5.0-rc1-Werror=all flag

stkw0, 2022-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I updated to esp-idf 5.0-rc1 and I saw that in this last version -Werror=all is enabled. This caused some compilation errors (in our code) that before were treated as errors. How can this flag be removed? I tried adding "-Wno-error" and "-Wno-error=all" without success. I know that I could manually add -Wno-error= for each type of errors but I would want to avoid this.

I thought that, being this an RC, maybe this was added for testing and wasn't removed later. Was this added on purpose?

igrr, 2022-11-21:
@stkw0 -Werror=all has been used in IDF for a very long time. For example, here it is [in release v3.3](https://github.com/espressif/esp-idf/blob/6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df/tools/cmake/idf_functions.cmake#L115). So there hasn't been any recent change regarding this flag.

> BTW the migration guide link on 5.0-rc1 release is broken, it returns a 404

Thank you for noticing. Have fixed the link.

stkw0, 2022-11-21:
Then I guess the problem is in some other place. What I can say is that the same code compiles correctly with 5.0-beta1 but not with 5.0-rc1. If needed I can provide a small test case

stkw0, 2022-11-21:
I modified the hello_world example to demonstrate the issue. Seems it only happens when a component is added. 
[Uploading build_test.zip…]()

stkw0, 2022-11-22:
The interesting thing is that the GCC version didn't change between 5.0-beta1 and 5.0-rc1, it's the same, 11.2.0. 

Also the implicit conversion from a C++ enum class that was possible with 5.0-beta1 is not possible anymore with 5.0-rc1. That's what made me think about the -Werror flag, in fact removing it produces the same behavior than with 5.0-beta1.

igrr, 2022-11-22:
> The interesting thing is that the GCC version didn't change between 5.0-beta1 and 5.0-rc1, it's the same, 11.2.0.

Correct, the version is the same in these releases. However the toolchain build we had in v5.0-beta1 was still configured with uint32_t == unsigned int. The newer toolchain build in v5.0-rc1 has uint32_t == unsigned long.

The format string type warning for unsigned long != unsigned int mismatch is produced also in the earlier versions of GCC, so that part didn't change.

stkw0, 2022-11-22:
I see,that makes sense, thank you very much.

closed 2022-11-22 by stkw0

apalazzi, 2025-01-10:
Hi,

I also have problems when compiling a custom module on my project under Eclipse, where compilation fails because -Werror=all is set.

Note that my issue has nothing to do with #9511 , I just want to disable the Werror=all flag.

ESP-IDV v5.1.5 because newer versions are n ot compatible with the Arduino library.