espressif/esp-idf #10214: Default WPA2 authmode causes PMK recalculation

boarchuz, 2022-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-4525-ge9c18ad02f

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

Custom ESP32

### Power Supply used.

Battery

### What is the expected behavior?

WPA2 PMK should only be recalculated when necessary.

### What is the actual behavior?

As of e9c18ad02f2f730c067372552853f9d8b7c7d40f, the WPA2 PMK is _always_ recalculated when reconnecting to the access point if threshold.authmode is zeroed, slowing down connection speed:

```
wifi_config_t wifi_config = {
    .sta = {
        .ssid = "MY_SSID",
        .password = "WPA2PASS",
    },
};
```
However, the PMK is correctly reused if threshold.authmode is set explicitly:
```
wifi_config_t wifi_config = {
    .sta = {
        .ssid = "MY_SSID",
        .password = "WPA2PASS",
        .threshold.authmode = WIFI_AUTH_WPA2_PSK,
    },
};
```

### Steps to reproduce.

Using the wifi/getting_started/station example:

Set your WPA2 SSID and password, and comment this line to zero the authmode:
https://github.com/espressif/esp-idf/blob/e9c18ad02f2f730c067372552853f9d8b7c7d40f/examples/wifi/getting_started/station/main/station_example_main.c#L119

Log/breakpoint here to be alerted when the PMK is recalculated:
https://github.com/espressif/esp-idf/blob/e9c18ad02f2f730c067372552853f9d8b7c7d40f/components/wpa_supplicant/src/rsn_supp/wpa.c#L2441-L2444

The PMK should only be calculated upon initial connection, and simply loaded from NVS on subsequent resets.

### More Information.

This is what I think is going on:

- Upon esp_wifi_set_config with threshold.authmode==0, the WiFi driver internally determines that this should be WIFI_AUTH_WPA2_PSK (3) and saves this modified value to NVS (key: `sta.minauth`). **Note: Prior to this commit, it saves the value 0, as provided.**
- When the WiFi driver is reinitialised, such as after restarting, it loads the config from NVS
- This time, when the exact same wifi_config_t is set, it does not match the loaded config because of the modified value, and an internal flag is set: `esp_wifi_sta_get_reset_param_internal()`
- wpa_set_passphrase recalculates the PMK based on this flag: https://github.com/espressif/esp-idf/blob/e9c18ad02f2f730c067372552853f9d8b7c7d40f/components/wpa_supplicant/src/rsn_supp/wpa.c#L2435

closed 2023-02-08 by espressif-bot

AxelLin, 2023-02-20:
@Aditi-Lonkar @kapilkedawat 
Just remind this issue was reported against v5.0 but the fix is not yet available in v5.0 branch.

AxelLin, 2023-03-14:
@Aditi-Lonkar Would you fix this for v5.0 branch?