espressif/esp-idf #10215: IDF v5.0-rc1 now requires DC pin to be valid conflicting with docs

atanisoft, 2022-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In ESP-IDF v4.4.x the esp_lcd component allows passing `GPIO_NUM_NC` as `esp_lcd_panel_io_spi_config_t :: dc_gpio_num`, with v5.0-rc1 this parameter now must map to a valid IO pin yet the code appears to still support `GPIO_NUM_NC`.

In the case of esp_lcd_touch which uses esp_lcd_panel_io_handle_t for talking to the touch controller this creates a problem since SPI based touch controllers (XPT2046, STMPE610, etc) will pick up the default value of GPIO_NUM_0 and potentially cause unintended pin state changes.

The new [comment](https://github.com/espressif/esp-idf/blob/v5.0-rc1/components/esp_lcd/src/esp_lcd_panel_io_spi.c#L69-L70) for the `dc_gpio_num` validation doesn't appear to be valid considering that all usages of `dc_gpio_num` still support `GPIO_NUM_NC`

suda-morris, 2022-11-22:
The limitation is removed in https://github.com/espressif/esp-idf/commit/7d3c9e9f3ef4adb68ad78053bd2404fdefd5e4b3
The commit has been backported to 5.0 internally, but still not synced to Github yet.

AxelLin, 2022-12-18:
v5.0 fix: e518e3efd3d75a7875e5db782a0213f1224fcdac

closed 2023-01-12 by Alvin1Zhang