espressif/esp-idf #10216: To exit from IDF monitor please use "Ctrl+]" not working

Willy0815, 2022-11-21:
I looked for the problem and found this issue, but meantime it's closed. For me the hints there didn't work. I'm using a Logitech Craft Keyboard with german layout, prepared to use with Windows and Apple Mac. I found that on my keyboard you have to 
type Strg (english Ctrl) and T (upper/lower case regardless) and release both keys. It's the beginning of an escape sequence. 
Then press T or X (again upper/lower regardless). This works. I misunderstand several hints in, that you have to press Ctrl,T,X all together. 
But that's the wrong way.

chipweinberger, 2022-11-22:
I've hit this before as well, during very fast boot loops.

dobairoland, 2022-11-22:
Yes, that is right. For those who has trouble with CTRL+], there is the possibility to open the menu with CTRL+T and then hit either CTRL+x or just x. 

But I personally recommend to switch to English keyboard in terminals where you are doing development.

closed 2022-11-22 by dobairoland