espressif/esp-idf #10219: how to support the spi+rgb lcd (st7701s)

Kingwulin, 2022-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

in idf5.0, there is a RGB lcd demo,but that cant driver the lcd IC st7701s,there are some doc about the lcd,how can I make it to support the lcd.
[Uploading ST7701S_SPEC_ V1.pdf…]()

Kingwulin, 2022-11-22:
> I guess you have to use the SPI interface to configure some ST7701 registers, then the normal RGB interface can work. FYI, Espressif has one reference design of RGB LCD kit which also relies on a special SPI initialization sequence: https://github.com/espressif/esp-dev-kits/blob/master/esp32-s3-lcd-ev-board/components/bsp/src/boards/lcd_panel_config.c

thank for your reply
anther question, how can I make it to running  in a idf5.0 demo derectly

suda-morris, 2022-11-23:
Except for the SPI initialization code, other parts you can reuse the IDF example code.

closed 2025-02-24 by suda-morris