espressif/esp-idf #10220: [FATFS] calling stat() on every file in a folder is too slow

chipweinberger, 2022-11-22:
### Is your feature request related to a problem?

**Goal:** get all paths in a folder that are directories

`stat()` is a very slow function.

Calling `stat()` on 300+ paths in the same directory takes 10+ seconds

But! calling *only* `readdir` on the same directory takes less than a second!

This is unfortunate because [in the code ](https://github.com/espressif/esp-idf/blob/7052a14d61d98dd387c4d99f435c42370b282fa6/components/fatfs/vfs/vfs_fat.c#L810), `vfs_fat_readdir_r` already has all the `stat()` information and just throws it away!

Is there anything I can do to speed this up?

### Describe the solution you'd like.

Not sure. 

Perhaps I can call `f_readdir` myself directly?

FRESULT f_readdir (
	FF_DIR* dp,			/* Pointer to the open directory object */
	FILINFO* fno		/* Pointer to file information to return */
)

chipweinberger, 2022-11-22:
So I may be pretty dumb! It looks like `dirent` has a `entry->d_type = DT_DIR` field that tells me exactly the information I need!

But, if I want to get filesize as well, perhaps I can do something like this:

```
int better_vfs_fat_readdir_r(void* ctx, DIR* pdir,
        struct dirent* entry, struct dirent** out_dirent, FILEINFO* out_fileinfo)
{
    assert(pdir);
    vfs_fat_dir_t* fat_dir = (vfs_fat_dir_t*) pdir;
    FRESULT res = f_readdir(&fat_dir->ffdir, &fat_dir->filinfo);
    if (res != FR_OK) {
        *out_dirent = NULL;
        ESP_LOGD(TAG, "%s: fresult=%d", __func__, res);
        return fresult_to_errno(res);
    }

    // copy fileinfo to output
    *out_fileinfo = fat_dir->filinfo;

    if (fat_dir->filinfo.fname[0] == 0) {
        // end of directory
        *out_dirent = NULL;
        return 0;
    }
    entry->d_ino = 0;
    if (fat_dir->filinfo.fattrib & AM_DIR) {
        entry->d_type = DT_DIR;
    } else {
        entry->d_type = DT_REG;
    }
    strlcpy(entry->d_name, fat_dir->filinfo.fname,
            sizeof(entry->d_name));
    fat_dir->offset++;
    *out_dirent = entry;
    return 0;
}

static struct dirent* better_vfs_fat_readdir(void* ctx, DIR* pdir, FILEINFO* out_fileinfo)
{
    vfs_fat_dir_t* fat_dir = (vfs_fat_dir_t*) pdir;
    struct dirent* out_dirent;
    int err = vfs_fat_readdir_r(ctx, pdir, &fat_dir->cur_dirent, &out_dirent);
    if (err != 0) {
        errno = err;
        return NULL;
    }
    return out_dirent;
}

main() {
  DIR *dir;
  struct dirent *entry;

  if ((dir = opendir("⁄")) == NULL)
    perror("opendir() error");
  else {
    puts("contents of root:");

    FILEINFO info;
    while ((entry = better_vfs_fat_readdir(dir, &info)) != NULL)
      printf("  %u\n", info->filesize);
    closedir(dir);
  }
}
```

igrr, 2022-11-22:
While it is unlikely we would add a non-standard API like better_vfs_fat_readdir, we can do an optimization for this specific case. `readdir` can cache the `FILINFO` structure for the current file and `stat` can peek into that cached structure. This will speed up the operation in the common case when you are calling `stat` for every directory entry, while iterating over the directory.

If implemented, this can close https://github.com/espressif/esp-idf/issues/9570 which I think was asking for a similar feature.

chipweinberger, 2022-11-22:
That's a clever solution. 

Yes, that would be a great idea.

chipweinberger, 2022-11-23:
When / if you implement caching, please consider adding it to the v4.4 branch as well. It'll be a massive improvement.

chipweinberger, 2022-11-24:
Just btw, to anyone that finds this thread, the code I posted does work! 

it is so much faster!

svofski, 2023-11-25:
In my project it can get up to 3-4 seconds per file just to query their size. This can't be right.

Thanks @chipweinberger for your workaround!

closed 2024-04-23 by espressif-bot

chipweinberger, 2024-12-18:
more people hitting this issue: https://github.com/mathieucarbou/ESPAsyncWebServer/discussions/165#discussioncomment-11576342