espressif/esp-idf #10221: regarding bluetooth data sending

beginner140BB, 2022-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Here i am having an issue of  bluetooth data sending, where i am sending the data only based on  the bluetooth exists using flag. where the flag is set to high on the event where there connection establishes.

i am settting the flag to high on the events
ESP_SPP_OPEN_EVT
ESP_SPP_SRV_OPEN_EVT
the initiator(or master or client) should check whether there exists a connection and if exists then send data based on the flag.
but its is entering into sniff mode

the below shown is the log
where master tries to send data based on flag

[image]

`while(bBTisConnected)
	{
	vTaskDelay(500);
	if(iSend_data_to_bluetooth("HEYSLAVE"))
	{
		printf("okay");
	}
	else
	{
		printf("not okay");
	}
	}`
	
	where bBTisConnected is high during ESP_SPP_OPEN_EVT on master side.
	
	
	
	anyone knows what may be the possible issue

beginner140BB, 2022-11-23:
> @beginner140BB Do you not understand why it enters sniff mode? Or, do you not want it to go into sniff mode at all?

dont want to go to sniff mode as
esp_spp_write() function will not work.

BetterJincheng, 2022-11-23:
@beginner140BB `vTaskDelay(500);` can be 5 seconds with 10ms-tick. When there is no data to be sent for a period time, it enters sniff mode to save power. If you take no care of power, you can disable sniff mode.
Currently, there is no such an API, we will add this API immediately.

beginner140BB, 2022-11-23:
> @beginner140BB `vTaskDelay(500);` can be 5 seconds with 10ms-tick. When there is no data to be sent for a period time, it enters sniff mode to save power. If you take no care of power, you can disable sniff mode. Currently, there is no such an API, we will add this API immediately.

actually i am sending a default string using esp_spp_write()

`int iSend_data_to_bluetooth(char *send_bt_data)
{
	esp_spp_write( handle,strlen(send_bt_data), (uint8_t *)send_bt_data);
	return 1;
}`

BetterJincheng, 2022-11-23:
@espressif-bot For a test, you can change `#define BTA_DM_LINK_POLICY_SETTINGS    (HCI_ENABLE_MASTER_SLAVE_SWITCH | HCI_ENABLE_HOLD_MODE | HCI_ENABLE_SNIFF_MODE | HCI_ENABLE_PARK_MODE)` to `#define BTA_DM_LINK_POLICY_SETTINGS    (HCI_ENABLE_MASTER_SLAVE_SWITCH | HCI_ENABLE_HOLD_MODE | HCI_ENABLE_PARK_MODE)`. And see if it works.

beginner140BB, 2022-11-23:
> 

reply from espressif

For the problem you are facing, please follow these steps to test:

1. Please download the “libbtdm_app.a”file in the attached and replace the “esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a” file under the esp-idf SDK that you used.

2. Delete the “build” file generated by your project and recompile it.

3. Please redownload the new compiled firmware for testing.

when we face issue in the case of slave device which accepts the connection
here issue is regarding master device when initiates the connection.

here we are able to initiate the connection but when we are sending data based on flag which enabled true during 
ESP_SPP_OPEN_EVT which comes when client connection opens i.e there is bluetooth connection based on the flag we are sending the data the flag is turning high but bluetooth is turning to sniff mode thereby not able to send data.

beginner140BB, 2022-11-23:
> @beginner140BB For a test, you can change `#define BTA_DM_LINK_POLICY_SETTINGS (HCI_ENABLE_MASTER_SLAVE_SWITCH | HCI_ENABLE_HOLD_MODE | HCI_ENABLE_SNIFF_MODE | HCI_ENABLE_PARK_MODE)` to `#define BTA_DM_LINK_POLICY_SETTINGS (HCI_ENABLE_MASTER_SLAVE_SWITCH | HCI_ENABLE_HOLD_MODE | HCI_ENABLE_PARK_MODE)`.
> 
> And see if it works.

it works its not entering to sniff mode
[image]

BetterJincheng, 2022-11-23:
@beginner140BB We will add an API to disable sniff mode.

beginner140BB, 2022-11-23:
> @beginner140BB We will add an API to disable sniff mode.

just curious will disabling sniff mode consumes more power.
and another question is that whether there is any other way to get to know the bluetooth connection exists other than using 
event information.

when i put while(1) its okay
but using flag read from event parameters its not okay

using simple while (1 )
4157) SPP_INITIATOR_DEMO: ESP_SPP_OPEN_EVT handle:129 rem_bda:[da:7a:fb:3f:00:00][0m
okay[0;32mI (4167) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT[0m

but when sending based on flag that set to high on open event the code is not executing the while loop

while(bBTisConnected)
`

     break;
    case ESP_SPP_OPEN_EVT:
        if (param->open.status == ESP_SPP_SUCCESS) {
        	bBTisConnected=1;
        	ESP_LOGI(SPP_TAG, "BT_connected:%d",bBTisConnected);

            ESP_LOGI(SPP_TAG, "ESP_SPP_OPEN_EVT handle:%d rem_bda:[%s]", param->srv_open.handle,
                     bda2str(param->srv_open.rem_bda, bda_str, sizeof(bda_str)));
            handle=param->srv_open.handle;
            /* Start to write the first data packet */
           // esp_spp_write(param->open.handle, SPP_DATA_LEN, spp_data);
           // send_bt_data("test",4);
           // s_p_data = spp_data;
            gettimeofday(&time_old, NULL);
        } else {
            ESP_LOGE(SPP_TAG, "ESP_SPP_OPEN_EVT status:%d", param->open.status);
        }
        break;
	

`

BetterJincheng, 2022-11-23:
@beginner140BB As I said before, if there is no data to be sent for a while, the link will go in sniff mode to save power. When `bBTisConnected` is set to high, it indicates the spp is open. Even though spp is open, it will still enter sniff mode when there is no data to send.
But in theory, even in sniff mode, calling `esp_spp_write()` can still exit sniff mode and send data successfully.
You can attach your complete code and I can check it for you.

beginner140BB, 2022-11-23:
[main.zip](https://github.com/espressif/esp-idf/files/10073292/main.zip)
this is the main code of initiator
could you please check this

BetterJincheng, 2022-11-23:
@beginner140BB In your `main.c`, `while (flag)` can be missed if flag is false and never come back...

beginner140BB, 2022-11-23:
> @beginner140BB In your `main.c`, `while (flag)` can be missed if flag is false and never come back...

oh no sorry
thanks for point out the simple issue.

closed 2022-11-23 by beginner140BB

BetterJincheng, 2022-11-24:
@beginner140BB Do you still use the SPP example?
You can use
`esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);`
to set the slave to non-connectable whenever you want.

beginner140BB, 2022-11-24:
> @beginner140BB Do you still use the SPP example? You can use `esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);` to set the slave to non-connectable whenever you want.

okay thanks for the support😊

BetterJincheng, 2025-12-24:
@ramesh12322 Does it work ?

ramesh12322, 2025-12-24:
> [@ramesh12322](https://github.com/ramesh12322) Does it work ?

@BetterJincheng After adding the API to the ESP-IDF source, I’m encountering build errors related to stdio.h. I suspect this may be due to a toolchain or environment configuration issue rather than the patch itself.I will reinstall ESP-IDF to ensure a clean environment and then test the new patch API. I’ll update my results after testing.

ramesh12322, 2025-12-31:
[main.zip](https://github.com/user-attachments/files/24390572/main.zip)

Sir, I kindly request you to please go through my code. If you find any issues or mistakes, please let me know.