espressif/esp-idf #10222: lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)

monurbstnc, 2022-11-22:
Hello,
When ı try to send a data via BLE using bluedroid/ble/gatt_server exeample. I got this error when try to conenct from linux machine.
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
What is the reason? How can I solve this?

xulongzhuang, 2022-11-22:
@monurbstnc 
From the esp32 log, the connection timed out, which may be caused by the device being far away or in a noisy environment.
[image]
The above is the command I sent to successfully connect with esp32 using gatttool.

You can also choose to use the bluetoothctl tool to try to connect to the esp32, enter `bluetoothctl` in the terminal to enter the command line interface, and enter the following commands in turn to connect to the esp32.
1. scan on
2. scan off
3. connect xx:xx:xx:xx:xx:xx

monurbstnc, 2022-11-22:
I try to increase transmit level of the ESP using this line of code
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV,ESP_PWR_LVL_P9);
And the esp is nearly 5 cm far from my pc but ı am getting the same error

[image]

monurbstnc, 2022-11-23:
> Has the linux machine been paired with the ESP device? If it has been paired, please delete the paired Bluetooth record on the system Bluetooth interface, erase the flash on the esp device and burn it again.
> Didnt paired . I erased the flashed and tried again but nothing changed.
> Are you able to connect to this ESP device using your phone's BLE app?
> After tried too many times ı connect
> Is it possible to use the packet capture tool to capture air packets?
> It is not
> I think you can still try to use the bluetoothctl tool to connect the ESP device, because in some Q&A it is found that bluetoothctl behaves differently from gatttool.
I tried with bluetoothctl but problem is stiil continue.
I noticed something:
I have nearly 10 esp32 development board (DEVKIT-V1). But I am able to  connect  just one of them. Do you think that the problem because of es32 board. If it is like that which one is more stable with BLE application.
By the way thanks for your helps . You helped me a lot :)

hyojincha, 2022-11-23:
I too am having the same problem.
This problem also appears on the Nimble gatt server.
Depending on your luck, when you reset the ESP, this happens or it works normally.
Also, every time it is reset, the delay at which Notify is received is very different.

ESP32 module : ESP32-WROOM-32D. 
IDF version :  v4.2.4-81-g04bf991178-dirty.

It's really serious. Please test it and take action.

Every time you reset with the same hardware and firmware in the same environment, the performance is different as shown below.
Notify is sent every 0.1 second.
Bad case, data is delayed for 0.5 seconds and then received at once.
Good case, uniformly data is received every 0.1 seconds.

Bad case
[image]

Good case
[image]

monurbstnc, 2022-11-23:
ESP32-D0WDQ6 (revision v1.0) chip is working fine on me but ESP32-D0WD-V3 (revision v3.0) chip is not working. I guess the problem is cause of chip

xulongzhuang, 2022-11-23:
`update connection params status = 0, min_int = 13, max_int = 32,conn_int = 30,latency = 0, timeout = 400`
The updated connection parameters reported by the above logs are successful. 0x13 means that the linux master actively disconnected the esp32 slave. The reason may be that the linux master does not support the updated connection parameters. You can set the connection parameters in the [ `esp_ble_gap_update_conn_params`](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L476) in the gatt_server code turn it up.

I have verified the following chips here: ESP32-D0WD (revision 0), ESP32-D0WD (revision 3), ESP32-D0WDQ6 (revision 0), they can all be connected normally after one or two failures, it seems that there is nothing the difference

What is the gatttool process you use, we can verify each other, the following is my connection process
```
$ sudo hcitool scan
$ suudo gattitool -b xx:xx:xx:xx:xx:xx -I
[xx:xx:xx:xx:xx:xx][LE]> connect
...
```

monurbstnc, 2022-11-23:
I just erase the flash then after upload (idf.py set-target esp32 / idf.py build / idf.py -p /dev/ttyUSB0 flash) the exact same code both ESP32-D0WD (revision 3) chip and ESP32-D0WDQ6 (revision 1). Only  chips that has ESP32-D0WDQ6 (revision v1.0) work good so far. 

I tried same connection process, but still same :(
-> suudo gatttool -b xx:xx:xx:xx:xx:xx -I
[image]

For connection parameters ı tried to change like that:

 conn_params.latency = 0;
 conn_params.max_int = 0x20; 
 conn_params.min_int = 0x0D;  
 conn_params.timeout = 400;    

but ıt didnt change anything.

monurbstnc, 2022-11-23:
> I just erase the flash then after upload (idf.py set-target esp32 / idf.py build / idf.py -p /dev/ttyUSB0 flash) the exact same code both ESP32-D0WD (revision 3) chip and ESP32-D0WDQ6 (revision 1). [...]

Can you just disconnect and try to connect again.  At first I tried maybe 4 or 5 times I connected but, when I disconnect I am not able to connect again.

xulongzhuang, 2022-11-23:
[image]
I tried using gatttool to repeatedly disconnect the ESP32, and it was still possible to connect after many times

monurbstnc, 2022-11-23:
I dont understand why ı cant make stable conenction between development board which has ESP32-D0WD (revision 3) inside it.

xulongzhuang, 2022-11-24:
Hi，@monurbstnc 
Are you referring above that two ESP32 boards cannot establish a connection?
For example, one ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_client example, and the other ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_server example, can the two still not establish communication? Do you have relevant logs to provide?

monurbstnc, 2022-11-25:
Are you referring above that two ESP32 boards cannot establish a connection? For example, one ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_client example, and the other ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_server example, can the two still not establish communication? Do you have relevant logs to provide?
Esps are communicating each other (one of them is ESP32-D0WD (revision 3) and the other ESP32-D0WDQ6 (revision 1) ) , but when ı try to connect esp32 via linux ESP32-D0WDQ6 (revision 1) chip works fine the other ESP32-D0WD (revision 3) chip is not working.

xulongzhuang, 2022-12-01:
> Are you referring above that two ESP32 boards cannot establish a connection? For example, one ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_client example, and the other ESP32 development board runs the examples/bluetooth/bluedroid/ble/gatt_server example, can the two still not establish communication? Do you have relevant logs to provide? Esps are communicating each other (one of them is ESP32-D0WD (revision 3) and the other ESP32-D0WDQ6 (revision 1) ) , but when ı try to connect esp32 via linux ESP32-D0WDQ6 (revision 1) chip works fine the other ESP32-D0WD (revision 3) chip is not working.
> 
> When you connnect with linux, did you use ESP32-D0WD (revision 3) chip ? I have nearly 10 esp32 devkt v1 7 of them have ESP32-D0WD (revision 3) chip which is not working for me and the others have ESP32-D0WDQ6 (revision 1) ) chip and working fine . Do you think that the problem is due to the development board. I mean maybe my devkits are fake or the design of the wroom-32's circuit is not good?

Yes, I am using the ESP32-D0WD (revision 3) chip, I am not sure about the differences in the development board, whether you can use the `esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9)` function to set the transmit power of the ESP32-D0WD (revision 3) chip Let's see if it works

NaveenKoyyada, 2023-02-08:
Hi, 
First of all, I think this forum which helped me to solve the BLE issues in the project I am working.

I too facing this problem, since todays I have been trying for root cause and possible solution.

IDE details: VScode, ESP_IDF v4.3
Module: ESP32 WROOM32 UE (8mb)

Below is the log for reference:
[image]

But interestingly, I checked with nRF IOS App I can able to provision the device and able to operate generic On/Off server which is implemented in my application.

Let me try **esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9)** .

Can any one help in this.

Thanks in Advance!

esp-cjh, 2024-10-23:
Hi, @NaveenKoyyada Has the problem been solved?