espressif/esp-idf #10223: FatFS error during ESP32 power OFF and power ON

simeondmr, 2022-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I'm using esp-idf 4.4.2 and a 16 MB WINBOND.
I have a strange problem using FatFS, for example, using this code:

    FILE *fp;
    char tmp[] = "test";

    while (true) {
        printf("TEST\n");
        fp = fopen("/root/test.txt", "w");
        if (fp == NULL) {
            ESP_LOGE("TEST", "Error creation file");
            break;
        }
        fwrite(tmp, 1, sizeof(tmp), fp);
        fclose(fp);
        ptr->deleteFile("/root/test.txt");
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }

If during the execution of this code, I turn off the ESP32 and then I turn on it, I have a sort of memory corruption(the free space of the winbond is zero and a lot of strange files are created):

I (507) Winbond: ** ERROR Directory Found: /root//��������.���
I (517) Winbond: ** ERROR Directory Found: /root//��������.���
I (527) Winbond: ** ERROR Directory Found: /root//��������.���
I (537) Winbond: ** ERROR Directory Found: /root//��������.���
I (547) Winbond: ** ERROR Directory Found: /root//��������.���
I (557) Winbond: ** ERROR Directory Found: /root//��������.���
I (567) Winbond: ** ERROR Directory Found: /root//��������.���
I (577) Winbond: ** ERROR Directory Found: /root//��������.���
I (587) Winbond: ** ERROR Directory Found: /root//��������.���
I (597) Winbond: ** ERROR Directory Found: /root//��������.���
I (607) Winbond: ** ERROR Directory Found: /root//��������.���
I (617) Winbond: ** ERROR Directory Found: /root//��������.���
I (627) Winbond: ** ERROR Directory Found: /root//��������.���
I (637) Winbond: ** ERROR Directory Found: /root//��������.���
I (647) Winbond: ** ERROR Directory Found: /root//��������.���
I (657) Winbond: ** ERROR Directory Found: /root//��������.���
I (667) Winbond: ** ERROR Directory Found: /root//��������.���
I (677) Winbond: ** ERROR Directory Found: /root//��������.���
I (687) Winbond: ** ERROR Directory Found: /root//��������.���
I (697) Winbond: ** ERROR Directory Found: /root//��������.���
I (707) Winbond: ** ERROR Directory Found: /root//��������.���
I (717) Winbond: ** ERROR Directory Found: /root//��������.���
I (727) Winbond: ** ERROR Directory Found: /root//��������.���
I (737) Winbond: ** ERROR Directory Found: /root//��������.���

Is a file system corruption?
What could be the problem?
How can I solve it, if the ESP will turn off during this operation?

Thanks

igrr, 2022-11-22:
@simeondmr Indeed, it seems like the filesystem gets corrupted. I'm afraid the only option you have is to implement some application-level checks and format the partition if the corruption happens. You may also consider using littlefs (available [as a component](https://components.espressif.com/components/joltwallet/littlefs)) which handles power-off scenarios better than FAT. Using NVS is also an option, it is power-failure-tolerant.

I have linked https://github.com/espressif/esp-idf/issues/10223 which is a feature request for adding power-off-tolerant filesystems in IDF. There is some work done on it already, but not ready to be released yet. You can click "subscribe" in that issue to get notified when this feature is ready.

simeondmr, 2022-11-22:
Hello @igrr, thanks as always.
I this that I will switch to littlefs, at the moment.

igrr, 2022-11-22:
Okay, I'll close this issue and will use #10223 for tracking the feature request for power-failure-tolerant filesystems in IDF.

closed 2022-11-22 by igrr