espressif/esp-idf #10227: How to get rtc time in risc-v？

meihualuo, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi
I want to get rtc time in ulp risc-v,try to do the following to get：

```
static uint64_t my_rtc_time_get(void)
{
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
    int attempts = 1000;
    ulp_riscv_delay_cycles(ULP_RISCV_CYCLES_PER_US * 100);
    uint64_t t;
    t = READ_PERI_REG(RTC_CNTL_TIME0_REG);
    t |= ((uint64_t)READ_PERI_REG(RTC_CNTL_TIME1_REG)) << 32;
    return t;
}
```
But this may cause an exception: Wakeup caused by COCPU crash.
what should I do?
idf version: ESP-IDF v5.1-dev-1883-g1657db56c9

meihualuo, 2022-11-23:
Maybe this is not the cause of the error, because I also used the i2c.

closed 2022-11-23 by meihualuo

2opremio, 2024-07-07:
I realized you are not checking the register to be valid `GET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_VALID) != 0`

And I see RTC_CNTL_TIME_VALID is not defined. Did you manage to get around this?

2opremio, 2024-07-08:
you can use `rtc_cntl_ll_get_rtc_time()`