espressif/esp-idf #10228: How to get the esp32c2 deep sleep time

Z-KY, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1751-gc09322430a-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP8684-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

I use the  IDF 'deep sleep' example project, change the sleep time to 10 second, expect the program to output the correct time.
The expected situation:
1.The print information is "Wake up from timer. Time spent in deep sleep: 10000ms".
2.Every time the chip wakeup from deep sleep,the result of function "time(NULL)" will add about to 10s.
[image]

[image]

### What is the actual behavior?

1.The printed sleep time is not the actual sleep time 10s.
2.timestamp is always 0.

### Steps to reproduce.

1. create the "deep sleep" example project from IDF.
2. change the sleep time to 10s.
3.when wakeup from RTC timer, print the result of function "time(NULL)".

### Debug Logs.

```plain
I (114) esp_image: segme�ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x167c
load:0x403acb70,len:0x96c
load:0x403aeb70,len:0x27d8
entry 0x403acb70
I (21) boot: ESP-IDF v5.1-dev-1751-gc09322430a-dirty 2nd stage bootloader
I (21) boot: compile time Nov 23 2022 14:38:43
I (22) boot: chip revision: v1.0
I (26) boot.esp32c2: MMU Page Size  : 32K
I (31) boot.esp32c2: SPI Speed      : 60MHz
I (36) boot.esp32c2: SPI Mode       : DIO
I (40) boot.esp32c2: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 230 lines cut ...]
I (163) cpu_start: Pro cpu start user code
I (163) cpu_start: cpu freq: 80000000 Hz
I (164) cpu_start: Application information:
I (166) cpu_start: Project name:     deep_sleep
I (171) cpu_start: App version:      1
I (176) cpu_start: Compile time:     Nov 23 2022 14:38:33
I (182) cpu_start: ELF file SHA256:  630e31d6de8cad18...
I (188) cpu_start: ESP-IDF:          v5.1-dev-1751-gc09322430a-dirty
I (195) cpu_start: Min chip rev:     v1.0
I (200) cpu_start: Max chip rev:     v1.99 
I (205) cpu_start: Chip rev:         v1.0
I (209) heap_init: Initializing. RAM available for dynamic allocation:
I (217) heap_init: At 3FCAB290 len 000318E0 (198 KiB): D/IRAM
I (223) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (231) spi_flash: detected chip: generic
I (234) spi_flash: flash io: dio
W (238) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (252) sleep: Configure to isolate all GPIO pins in sleep state
I (258) sleep: Enable automatic switching of GPIO sleep configuration
I (265) cpu_start: Starting scheduler.
Wake up from timer. Time spent in deep sleep: 4199ms
timestamp: 0
Enabling timer wakeup, 10s
Entering deep sleep
```

igrr, 2022-11-23:
Unfortunately there currently is no solution for persisting the system time after deep sleep on C2, as the chip doesn't have any RTC memory. You'd probably have to use an external RTC chip or switch to C3.

Z-KY, 2022-11-23:
> Unfortunately there currently is no solution for persisting the system time after deep sleep on C2, as the chip doesn't have any RTC memory. You'd probably have to use an external RTC chip or switch to C3.

So every time the chip enter deep sleep mode, the function "time(NULL)" will lose effect?

igrr, 2022-11-23:
Correct, you need to get the time from SNTP or an external RTC after power-on.

Z-KY, 2022-11-23:
> Correct, you need to get the time from SNTP or an external RTC after power-on.

Thankyou~

ESP-Marius, 2023-04-27:
f8cc2ec86daf9e14743886ed530fc902bdd9ec1f added a feature for storing the time in one of the few RTC registers that the chips has, so now it should be possible to have a persistent system time even after deep sleep.

closed 2023-04-27 by ESP-Marius