espressif/esp-idf #10231: receive buffer from  aws shadow too small to hold string value.

sandipmaurya, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]

I am sending form string from aws shadow

void windowActuate_Callback_str(const char *pJsonString, uint32_t JsonStringDataLen, jsonStruct_t *pContext)
{
    JsonStringDataLen = 15;
    IOT_UNUSED(pJsonString);
    IOT_UNUSED(JsonStringDataLen);
    
    //printf("length is %d\n",JsonStringDataLen);

    if (pContext != NULL)
    {
        ESP_LOGI(TAG, "Delta - Window state changed to (string) %c",  *(char *)(pContext->pData));
    }
}

0xjakob, 2022-11-25:
@sandipmaurya is the problematic code located in ESP-IDF? Why can't you increase the buffer size?

sandipmaurya, 2022-11-25:
Thank you for replying. now it is working.
I assigned size in  .dataLength same as string whatever i want to assign data

   char device_local_timezone[12] = "GMT+5:30";
    jsonStruct_t device_local_timezone_h;
    device_local_timezone_h.cb = windowActuate_Callback_str;
    device_local_timezone_h.pData = &device_local_timezone;
    device_local_timezone_h.pKey = "device_local_timezone";
    device_local_timezone_h.type = SHADOW_JSON_STRING;
    device_local_timezone_h.dataLength = _sizeof(device_local_timezone);_

closed 2022-11-25 by sandipmaurya