espressif/esp-idf #10232: Does ESP32 support ATECC608B secure element I2C communication with 400 000 baudrate ?

LumbirBwut, 2022-11-23:
### Is your feature request related to a problem?

Hi,
I'm actually using an esp32 and communicating with my ATECC608B cryptochip through I2C.
Everything is working fine using standard configuration, however as soon as I increase the baudrate to 400 000, I receive only I2C fail communication with my cryptochip.

### Describe the solution you'd like.

I'll like to be able to just set my I2C baudrate to 400 000 as the secure element seems to be able to support it.

### Describe alternatives you've considered.

Looking at this file (https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/hal/hal_esp32_i2c.c), it looks like the baud rate on the bus is somehow caped:
```c
ATCA_STATUS hal_i2c_init(ATCAIface iface, ATCAIfaceCfg *cfg)
{
    esp_err_t rc = ESP_FAIL;
    int bus = cfg->atcai2c.bus;

    if (bus >= 0 && bus < MAX_I2C_BUSES)
    {
        if (0 == i2c_hal_data[bus].ref_ct)
        {
            i2c_hal_data[bus].ref_ct = 1;
            i2c_hal_data[bus].conf.mode = I2C_MODE_MASTER;
            i2c_hal_data[bus].conf.sda_pullup_en = GPIO_PULLUP_DISABLE;
            i2c_hal_data[bus].conf.scl_pullup_en = GPIO_PULLUP_DISABLE;
            i2c_hal_data[bus].conf.master.clk_speed = 100000; //cfg->atcai2c.baud;
```

### Additional context.

I already open an Issue to discuss this with Microchip (https://github.com/MicrochipTech/cryptoauthlib/issues/318) but they redirect me to you.

Thanks in advance 👍

AdityaHPatwardhan, 2023-01-04:
Hi @LumbirBwut sorry for the delayed reply.
I think currently the baud rate is hard-coded to `100000`.

I think just changing that value to `400000` won't fix the issue.
Please update the value [here](https://github.com/espressif/esp-cryptoauthlib/blob/2c265deeb1a37040337c6de330421f68c9f6d6bd/port/atca_cfgs_port.c#L50) as well.

I have provided a patch that does the same changes as mentioned above. I think this should fix your issue.

[baud_rate.patch.zip](https://github.com/espressif/esp-idf/files/10342201/baud_rate.patch.zip)

Thanks and regards

AdityaHPatwardhan, 2023-01-24:
@LumbirBwut  Your issue should be fixed with https://github.com/espressif/esp-cryptoauthlib/commit/3d3583b0649719580241c3d18c49bec59d242f5c

I am closing the issue, please feel free to re-open if required.

closed 2023-01-24 by AdityaHPatwardhan