espressif/esp-idf #10233: Nimble characteristics.access_cb conn_handle Argument Invalid ( Always is set to 0xFFFF)

Stoician, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-MINI-1 on a custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The documentation on the NimBLE website details the expected values given to the `access_cb` of a characteristic.
https://mynewt.apache.org/latest/tutorials/ble/bleprph/bleprph-sections/bleprph-chr-access.html

"conn_handle: Indicates which connection the characteristic access was sent over."

So if multiple centrals are connected to the ESP32C3, then conn_handle argument of the access_cb can be used to distinguish between them. It would be expected that the `access_cb` `conn_handle` would match the `event->[event type].conn_handle` argument of the advertisement callback assigned with the `ble_gap_adv_start` function.

### What is the actual behavior?

During a read, the `conn_handle` argument of the `access_cb` is always 0xFFFF regardless of whether one or multiple centrals are connected to the ESP32C3.

The `ctxt` argument of `access_cb` is valid and sends data to the central that is requesting it.

The `event->[event type].conn_handle` argument of the advertisement callback is correct.

So it is just the `conn_handle` argument of the `access_cb` that is wrong.

### Steps to reproduce.

access_cb:
```
static int ble_svc_gatt_read_handler(uint16_t conn_handle, uint16_t attr_handle, struct ble_gatt_access_ctxt *ctxt, void *arg) {
    ESP_LOGI(tag, "handle %04X", conn_handle); // Always returns 0xFFFF

   // ... application code to copy data into ctxt->om which is working fine as long as it ignores conn_handle

    return 0;
}
```

advertisement callback:
```
static int ble_spp_server_gap_event(struct ble_gap_event *event, void *arg)
{
    struct ble_gap_conn_desc desc;
    int rc;

    switch (event->type)
    {

    case BLE_GAP_EVENT_CONNECT:
        MODLOG_DFLT(INFO, "connection %s; status=%d ",
                    event->connect.status == 0 ? "established" : "failed",
                    event->connect.status);
        // code that uses conn_handle without any issues
        if (event->connect.status == 0) {
            ble_gap_conn_find(event->connect.conn_handle, &desc);
            ble_spp_server_print_conn_desc(&desc);
            
            if (!ble_handle_exists(event->connect.conn_handle)) {
                ble_handle_add(event->connect.conn_handle);
            }
            
            handleConnectionStateChanges(event->connect.conn_handle);
        }
        // check to make sure advertising continues so another device can connect
        ble_app_advertise();
        return 0;

   // ... handling for additional event types. these also use event->[event type].conn_handle without issue. 
   // That is, conn_handle is consistent over processing multiple events for each connected central.

    }
}
```

```
static void ble_app_advertise() {
    if (!ble_gap_adv_active()) {
        struct ble_gap_adv_params adv_params = {0};

        handleConnectionStateChanges(0);
        
        ble_homeaware_set_adv_data((ble_uuid128_t*)&ha_uuid128, 1);
        
        adv_params.conn_mode = BLE_GAP_CONN_MODE_UND;
        adv_params.disc_mode = BLE_GAP_DISC_MODE_GEN;
        
        ble_gap_adv_start(BLE_OWN_ADDR_RPA_RANDOM_DEFAULT, NULL, BLE_HS_FOREVER, &adv_params, ble_spp_server_gap_event, NULL);
    }
}
```
```
int ble_homeaware_set_adv_data(ble_uuid128_t *uuid128, uint16_t version) {

    struct ble_hs_adv_fields fields = {0};
    int rc;

    

    fields.uuids128 = uuid128;
    fields.num_uuids128 = 1;
    fields.uuids128_is_complete = 1;

    fields.mfg_data = mfg;
    fields.mfg_data_len = 3;

    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;
    fields.tx_pwr_lvl_is_present = 1;

    /* Advertise two flags:
     *     o Discoverability in forthcoming advertisement (general)
     *     o BLE-only (BR/EDR unsupported).
     */
    fields.flags = BLE_HS_ADV_F_DISC_GEN |
                   BLE_HS_ADV_F_BREDR_UNSUP;

    rc = ble_gap_adv_set_fields(&fields);
    
    fields.name = (uint8_t *) BLE_NAME;
    fields.name_len = 9;//strlen(BLE_NAME);
    fields.name_is_complete = 1;

    fields.uuids128 = NULL;
    fields.num_uuids128 = 0;
    fields.uuids128_is_complete = 0;

    fields.mfg_data = NULL;
    fields.mfg_data_len = 0;

    fields.tx_pwr_lvl = 0;
    fields.tx_pwr_lvl_is_present = 0;

    fields.flags = 0;
    
    ble_gap_adv_rsp_set_fields(&fields);
    
    return rc;
}
```

SumeetSingh19, 2022-11-28:
Hi @Stoician , can you share the application that you observed this behaviour in.

SumeetSingh19, 2022-11-28:
Also share what device you observed this behavior in.

Stoician, 2022-11-29:
Attached is a stripped down version of the code that doesn't interact with any io pins, so it should be able to run on the ESP32-C3-DevKitM-1.
[BT_ESP_Simple.zip](https://github.com/espressif/esp-idf/files/10117363/BT_ESP_Simple.zip)
To quickly use, you can install the HomeAware app from the store for either Android or iPhone.
After connection nothing will happen.
As can be seen in the esp log output conn_handle is always 0xFFFF.
If line 578 "//k = 0;" in main.c is uncommented conn_handle will be ignored. Then after reconnection, a Door event should trigger on the phone every few seconds.

SumeetSingh19, 2022-11-30:
Hi @Stoician , can you elaborate how you connected the HomeAware app to your application?

SumeetSingh19, 2022-12-05:
Hi @Stoician ,
I tried running your app on ESP32C3 with Idf v4.4.3. It starts advertising itself as HomeAware, so I tried to connect the Home Aware app to it but the pairing fails.
Could you please tell me the steps to reproduce this issue?

Stoician, 2022-12-05:
I made some assumptions.
There was still some stale data on my module that was making the code in the zip file work.
When I switched to the devkit it stopped working. Although, for me it still connected, but the door did not come when k=0.
That was for Android. For the iPhone version the connection and door event come through. I didn't develop the apps, there must be some difference between them.

So here are a few things to get consistent results.

Use updated file, I removed the test flags that would result in a compile error before and made a few other tweaks:
[BT_ESP_Simple2.zip](https://github.com/espressif/esp-idf/files/10156074/BT_ESP_Simple2.zip)

Do not use the homeaware app.
Instead use "nRF Connect" when on android. 
After scan and connecting to the module, go to the devices tab.
In the HomeAware Service, for the first Characteristic that starts with "81eb77bd" press the single down arrow. It may ask to pair at this point. If so, press the down arrow again. The Value should become "m1005;" if k=0.

Another issue with the devkit. Not all log data is sent through the UART port. I'm not really sure why, it did before when I used the devkit a few months ago. But if you switch the resistors on the devkit to use the ESP32's built in USB port, then all of the log messages can be seen from it's virtual com port.

Once I got to this point, I discovered something else. conn_handle is correct when the read is initiated by pressing the single down arrow. But if notifications are turned on (by pressing the clustered down arrows in the same characteristic), then conn_handle is 0xFFFF when the read is triggered by the notification.

I realized that I had made some modifications to nimble, so I reverted to nimble-1.3.0-idf which I think is the version pointed to by idf v4.4.3. The symptoms as described in the above paragraph still persisted.

Stoician, 2023-06-13:
It looks like the nimble team isn't interested in this problem, but for those that want to be able to make sure the correct data is getting to the correct `conn_handle`, I found a better way.

It is possible to send data directly and at the same time as triggering the notification. Doing it this way skips the characteristic callback. That is done by calling `ble_gattc_notify_custom` directly.

`struct os_mbuf *txom = ble_hs_mbuf_from_flat(data, len);`
`ble_gatts_notify_custom(conn_handle, chr, txom);`

Where
`data` is the array you want to send to a specific phone.
`conn_handle` is the handle of a specific phone you want to send the data.
`chr` is the characteristic handle that was registered for notification by the phone.

Since `ble_gatts_notify_custom` also triggers the notification, there is no need to use `ble_gatts_chr_updated` and keep track of which phone is which in the read callback. So now the read callback will only be called if a phone does a read directly, which as mentioned above does return the correct conn_handle.

Singh-Sumeet, 2024-03-05:
Hi @Stoician ,
> conn_handle: Indicates which connection the characteristic access was sent over.

This is true, however, 0xFFFF is a valid value for conn_handle. Let me explain why.

When a remote device read/writes a value to a characteristic, the `access_cb` will have it's connection handle, to indicate which connection sent the request. 
But sometimes the local stack itself reads/writes to a characteristic, in that case we cannot put a value for the connection handle (0xFFFF represents that the read/write request was not received from a remote device, but from the local stack itself).

In your case, this is being triggered by `ble_gatts_chr_updated`. This API's role is to send a notification to the remote device, and the value of the notification is the value of the characteristic itself. Hence, the API sends a read request to the characteristic. Note that this request was sent by the local stack (in order to fill the notification), hence 0xFFFF is received (since no remote device is involved in this request).

Hence, it is suggested that if 0xFFFF is received, the request is handled as normal, to allow the ongoing notification procedure to complete.

If you have questions regarding this explanation, please reach out.

SumeetSingh19, 2024-04-30:
Closing issue. Feel free to reopen if further questions.

closed 2024-04-30 by rahult-github