espressif/esp-idf #10236: CDC ACM transmit problem

beckmx, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-rc1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

Creating a new host usb device I am able to receive data from cp210x and ftdi, however when I try to send information I get an abort() failure regarding the transmit feature. The expectation is that I should be able to transmit without failures

### What is the actual behavior?

An abort() happens and is not able to transmit the data

### Steps to reproduce.

1. Compile the sample project cdc_acm_vcp
2. In the method for handling RX, write this code
```
static void handle_rx(uint8_t *data, size_t data_len, void *arg)
{
    printf("%.*s", data_len, data);
    ESP_ERROR_CHECK(vcp->set_control_line_state(false, true));
    static char text2[] = "test_string\r\n";
    ESP_ERROR_CHECK(vcp->tx_blocking((uint8_t *)text2, strlen(text2)));
}
```
4. Run the project and expectation is whenever a new message is received, anotherone is sent
5. abort() because of failure

### Debug Logs.

```plain
.[0;32mI (29) boot: ESP-IDF v5.0-rc1-4-g919b3e621e 2nd stage bootloader.[0m
.[0;32mI (29) boot: compile time 18:17:09.[0m
.[0;32mI (29) boot: chip revision: v0.1.[0m
.[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0.[0m
.[0;32mI (39) boot.esp32s3: Boot SPI Speed : 80MHz.[0m
.[0;32mI (44) boot.esp32s3: SPI Mode       : DIO.[0m
.[0;32mI (49) boot.esp32s3: SPI Flash Size : 2MB.[0m
.[0;32mI (54) boot: Enabling RNG early entropy source....[0m
.[0;32mI (59) boot: Partition Table:.[0m
.[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length.[0m
.[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000.[0m
.[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000.[0m
.[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00100000.[0m
.[0;32mI (92) boot: End of partition table.[0m
.[0;32mI (97) boot_comm: chip revision: 1, min. application chip revision: 0.[0m
.[0;32mI (104) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0d158h ( 53592) map.[0m
.[0;32mI (122) esp_image: segment 1: paddr=0001d180 vaddr=3fc92100 size=02b88h ( 11144) load.[0m
.[0;32mI (124) esp_image: segment 2: paddr=0001fd10 vaddr=40374000 size=00308h (   776) load.[0m
.[0;32mI (129) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=24ea0h (151200) map.[0m
.[0;32mI (165) esp_image: segment 4: paddr=00044ec8 vaddr=40374308 size=0dde8h ( 56808) load.[0m
.[0;32mI (178) esp_image: segment 5: paddr=00052cb8 vaddr=50000000 size=00010h (    16) load.[0m
.[0;32mI (184) boot: Loaded app from partition at offset 0x10000.[0m
.[0;32mI (184) boot: Disabling RNG early entropy source....[0m
.[0;32mI (198) cpu_start: Pro cpu up..[0m
.[0;32mI (198) cpu_start: Starting app cpu, entry point is 0x40375320.[0m
.[0;32mI (0) cpu_start: App cpu up..[0m
.[0;32mI (212) cpu_start: Pro cpu start user code.[0m
.[0;32mI (212) cpu_start: cpu freq: 160000000 Hz.[0m
.[0;32mI (212) cpu_start: Application information:.[0m
.[0;32mI (215) cpu_start: Project name:     cdc_acm_vcp.[0m
.[0;32mI (221) cpu_start: App version:      1.[0m
.[0;32mI (225) cpu_start: Compile time:     Nov 22 2022 18:16:58.[0m
.[0;32mI (231) cpu_start: ELF file SHA256:  638fed8a2c608a63....[0m
.[0;32mI (237) cpu_start: ESP-IDF:          v5.0-rc1-4-g919b3e621e.[0m
.[0;32mI (243) heap_init: Initializing. RAM available for dynamic allocation:.[0m
.[0;32mI (251) heap_init: At 3FC95710 len 00054000 (336 KiB): D/IRAM.[0m
.[0;32mI (257) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM.[0m
.[0;32mI (264) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM.[0m
.[0;32mI (270) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM.[0m
.[0;32mI (277) spi_flash: detected chip: gd.[0m
.[0;32mI (280) spi_flash: flash io: dio.[0m
.[0;33mW (284) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header..[0m
.[0;32mI (298) cpu_start: Starting scheduler on PRO CPU..[0m
.[0;32mI (0) cpu_start: Starting scheduler on APP CPU..[0m
.[0;32mI (318) VCP example: Installing USB Host.[0m
.[0;32mI (348) VCP example: Installing CDC-ACM driver.[0m
.[0;32mI (348) VCP example: Opening CH340 device.[0m
.[0;32mI (748) FT23x: Baudrate required: 115200, set: 115384.[0m
.[0;32mI (748) VCP example: Setting up line coding.[0m
.[0;32mI (748) FT23x: Baudrate required: 115200, set: 115384.[0m
hello world:12000

.[0;31mE (5938) cdc_acm: cdc_acm_reset_transfer_endpoint(242): .[0m
.[0;31mE (5938) FT23x: set_control_line_state(85): .[0m
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x42008530
file: "./main/cdc_acm_vcp.cpp" line 34
func: void handle_rx(uint8_t*, size_t, void*)
expression: vcp->set_control_line_state(false, true)

abort() was called at PC 0x40379d53 on core 0

Backtrace: 0x40375e22:0x3fcec160 0x40379d5d:0x3fcec180 0x4038012e:0x3fcec1a0 0x40379d53:0x3fcec210 0x42008530:0x3fcec240 0x42023e55:0x3fcec260 0x42008f0d:0x3fcec290 0x4200a3ac:0x3fcec2b0 0x4200ace7:0x3fcec2d0 0x42008cf6:0x3fcec300 0x4037cf39:0x3fcec320
```

### More Information.

I tried this from idf 5.0 and idf 4.4.3 both fail with the same output

beckmx, 2022-11-24:
I tested and yes, you were right, I had to create another task to control the usb states, thanx :)

beckmx, 2022-11-24:
ok I noticed something wierd now, if I move the connection part into a RTOS task the try, catch doesnt work anymore and aborts the application. In the main app I created this method:

`xTaskCreate(usb_communications, "usb_comm", 8192, NULL, 2, NULL);`

and in the task I have this:

```
void usb_communications(void *arg)
{
    while (1) {
       const cdc_acm_host_device_config_t dev_config = {
            .connection_timeout_ms = 10000,
            .out_buffer_size = 64,
            .event_cb = handle_event,
            .data_cb = handle_rx,
            .user_arg = NULL,
        };

        
        try {
            Serial.println("Opening device");
            vcp = FT23x::open_ftdi(FTDI_FT232_PID, &dev_config);//<----Looks like it breaks here
        } catch (esp_err_t err) {
            Serial.println( "The required device was not opened.\nExiting...");
            usbConnected = false;
            return;
        }

        Serial.println( "Setting up line coding");
        cdc_acm_line_coding_t line_coding = {
            .dwDTERate = EXAMPLE_BAUDRATE,
            .bCharFormat = EXAMPLE_STOP_BITS,
            .bParityType = EXAMPLE_PARITY,
            .bDataBits = EXAMPLE_DATA_BITS,
        };
        ESP_ERROR_CHECK(vcp->line_coding_set(&line_coding));

        xSemaphoreTake(device_disconnected_sem, portMAX_DELAY);
        delete vcp;
    }
}
```

The stack trace is this one:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376e14
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
Waiting a client connection to notify...
Installing USB Host
Installing CDC-ACM driver
Opening device

abort() was called at PC 0x42080b7f on core 0

Backtrace:0x4037720a:0x3fca51b00x4037d6e9:0x3fca51d0 0x403836e5:0x3fca51f0 0x42080b7f:0x3fca5270 0x42080bc6:0x3fca5290 0x42080163:0x3fca52b0 0x420031ce:0x3fca52d0 0x42003201:0x3fca52f0 0x4200345a:0x3fca5310 
```

beckmx, 2022-11-24:
Found the answer, looks like a top task has problems, it needs "an inner" method to run the code, I found the answer from an old issue https://github.com/espressif/esp-idf/issues/2300

closed 2022-11-24 by beckmx

nucleon, 2023-02-07:
@tore-espressif may I ask where in the docs you provided clearer instructions on how to signal your application that it can send data? I'm currently having the same issue, I'm quite new to the esp-idf event structure.