espressif/esp-idf #10238: parttool.py crashes when trying to write a nvs partition

oresk, 2022-11-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-rc1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32-s3-devkit

### Power Supply used.

External 3.3V

### What is the expected behavior?

So I have just migrated from fa9cc49b9e4614b3d6412f0480cadf30a9ad650d to the esp-idf v5.0-rc1 release. 

Have been using this command to flash the nvs partition

`parttool.py write_partition --partition-name=nvs --input=parameters/parameters_xxx.bin`

And the partition would be flashed.

### What is the actual behavior?

With the current v5.0rc1 release I get this error when trying to run the same command 

`
Traceback (most recent call last):
  File "/home/lovro/esp/esp-idf-v5.0-rc1/components/partition_table/parttool.py", line 358, in <module>
    main()
  File "/home/lovro/esp/esp-idf-v5.0-rc1/components/partition_table/parttool.py", line 324, in main
    if args.extra_partition_subtypes:
AttributeError: 'Namespace' object has no attribute 'extra_partition_subtypes'. Did you mean: 'partition_subtype'?
`

Commenting out the lines 324 and 325 gets it going again.

I don't really understand what the extra_partition_subtypes should do.

### Steps to reproduce.

1. Make a nvs partition
2. Try to use the parttool.py to write the partition
3. Get the error :-)

### Debug Logs.

```plain
`
Traceback (most recent call last):
  File "/home/lovro/esp/esp-idf-v5.0-rc1/components/partition_table/parttool.py", line 358, in <module>
    main()
  File "/home/lovro/esp/esp-idf-v5.0-rc1/components/partition_table/parttool.py", line 324, in main
    if args.extra_partition_subtypes:
AttributeError: 'Namespace' object has no attribute 'extra_partition_subtypes'. Did you mean: 'partition_subtype'?
`
```

Harshal5, 2022-11-24:
Hello @oresk
The issue has been fixed in the commit `aca27c8db23a3afb4e65de0f8e38e6b17f2262d4` on `master` branch.
Can you cherry-pick the commit until we back-port it into `release/v5.0-rc1`?

oresk, 2022-11-24:
Works now!

Thnx for the quick response

closed 2022-11-24 by oresk