espressif/esp-idf #10239: Question about esp_gap_ble_set_channels.

hyojincha, 2022-11-24:
in esp_gap_ble_api.h 

esp_err_t esp_gap_ble_set_channels(esp_gap_ble_channels channels);
channels :   The n th such field (in the range 0 to 36) contains the value for the link layer channel index n.

It is specified.

In the BLE spec, the order of channels is from 2402Mhz to 37,0,1,2,3,4,5,6,7,8,9,10,38,11,12,13, ...,39.

1. Do the arguments of esp_gap_ble_set_channels also follow this order? Or is the order 0,1,2,3.. ?

2. Does each array mean the channel with the lowest MSB or vice versa?

for example

esp_gap_ble_channels channel={0x7f,0xff,0xff,0xff,0xff};

, is channel 0 (or 37) set as a bad channel?

xiewenxiang, 2022-11-24:
The size of `channels` is 5 Octets (37 Bits meaningful), The following is a description of the agreement:

```c
This parameter contains 37 1-bit fields.
The nth such field (in the range 0 to 36) contains the value for the link layer channel index n.
Channel n is bad = 0. Channel n is unknown = 1.
The most significant bits are reserved and shall be set to 0for future use.
At least one channel shall be marked as unknown.
```

The `channels[0]` indicates channels 0 to 7 and the `channels[4]` indicates channel 32 to 39.

For example:

`channels[3] = 0xF0` means channel 28 29 30 31 are used.

Note:

Channel 37 38 39 should not be used,

hyojincha, 2022-11-24:
> thank you Let me ask you one more question.
Does channels[0] = 0x0F use channels 0 1 2 3 ?
Or are you using 37 0 1 2 ?

hyojincha, 2022-11-24:
> than you sir.

closed 2022-11-24 by hyojincha