espressif/esp-idf #10240: Bug in nvs C++ example

ljden, 2022-11-24:
C++ NVS example stores the error value in `result` but checks the value of `err`
https://github.com/espressif/esp-idf/blob/1b16ef6cfc2479a08136782f9dc57effefa86f66/examples/storage/nvs_rw_value_cxx/main/nvs_value_example_main.cpp#L35-L38

As the rest of the code reuses `err`, this section should also reuse `err`. If there's some reason we need a separate variable then the correct variable should be tested
```diff
     esp_err_t err = nvs_flash_init();
...
-    esp_err_t result;
     // Handle will automatically close when going out of scope or when it's reset.
-    std::shared_ptr<nvs::NVSHandle> handle = nvs::open_nvs_handle("storage", NVS_READWRITE, &result);
+    std::shared_ptr<nvs::NVSHandle> handle = nvs::open_nvs_handle("storage", NVS_READWRITE, &err);
     if (err != ESP_OK) {
```

0xjakob, 2022-11-25:
@ljden Thanks for noticing this! We'll fix it.

ljden, 2022-11-29:
@0xjakob I noticed that the `open_nvs_handle()` function returns a `unique_ptr` but the comment in the header states that it should return a shared pointer - and the example code stores the return val in a shared pointer. Am I missing something or is this a mistake? If so, I can open a new ticket

0xjakob, 2022-11-29:
@ljden 
> Am I missing something or is this a mistake?

It's a documentation mistake. The `unique_ptr` is intentional because `unique_ptr` doesn't allocate memory, hence can't throw an exception if that happens. Users can choose if they want to use `shared_ptr` or `unique_ptr`. I can fix it within this ticket.

closed 2022-12-22 by espressif-bot