espressif/esp-idf #10243: SYS_ARCH_UNPROTECT -> Task watchdog got triggered: moving from 4.4 to 5.1-dev

surfskidude, 2022-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1908-g439a709c42

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

v5.1-dev-1908-g439a709c42

### Power Supply used.

USB

### What is the expected behavior?

I just ported my app from 4.4 to 5.1 and I now get the following printouts each time my code calls SYS_ARCH_UNPROTECT

E (131517) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (131517) task_wdt:  - IDLE (CPU 1)
E (131517) task_wdt: Tasks currently running:
E (131517) task_wdt: CPU 0: IDLE
E (131517) task_wdt: CPU 1: main.c
E (131517) task_wdt: Print CPU 1 backtrace

Stack after SYS_ARCH_UNPROTECT

0x40086846: esp_crosscore_isr at /esp/esp-idf/components/esp_system/crosscore_int.c:96
0x400835ed: _xt_lowint1 at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1121
0x40092109: vPortClearInterruptMaskFromISR at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:574
 (inlined by) vPortExitCritical at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:514
0x4008edc5: xQueueGenericSend at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:949
0x401daa60: sys_mutex_unlock at /esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:102
 (inlined by) sys_arch_unprotect at /esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:510

This worked fine in 4.4 (and other lwIP based RTOSs). I have protected code as follows:
         SYS_ARCH_DECL_PROTECT(lev);
         SYS_ARCH_PROTECT(lev);
         my protected code;
         SYS_ARCH_UNPROTECT(lev);

### What is the actual behavior?

see above

### Steps to reproduce.

see above

surfskidude, 2022-12-02:
Some more info: I am running a web server and when the web server is idle, I get the watchdog error from the IDLE process as explained above. I created a basic shell script that fetches a web page every second and I no longer get the error. In short, this must be a bug in the IDLE process.

AlbertShown, 2023-02-08:
I have the same error for the SPI motors driver, if the motor is not connected, `spi_device_transmit()` triggers the error (IDLE).