espressif/esp-idf #10245: restart immediately when enter deep sleep

robbietree8, 2022-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## basic information
chip: esp32-c3
esp-idf: release/v4.4 commit-id: 76caa64286b0e93207ad06f68830a223be49cc88

## behavior & help wanted

I'm using gpio4 as the wake up source, at the same time, gpio4 is also the button.
When long press gpio4, it will trigger to deep sleep.
What confusing me is that the system will boot immediately after deep sleep.

Is there any config help to avoid it?

### code

```c
#define FEATURE_TOGGLE_PIN 4

static void button_long_press_hold_cb(void *arg1, void *arg2)
{
    ESP_LOGI(TAG, "button long press");
    enter_deep_sleep();
}

static void setup_gpio_button()
{
    ESP_LOGI(TAG, "gpio button init on pin %d", FEATURE_TOGGLE_PIN);
    button_config_t cfg = {
        .type = BUTTON_TYPE_GPIO,
        .gpio_button_config = {
            .gpio_num = FEATURE_TOGGLE_PIN,
            .active_level = 1,
        },
    };
    button_handle_t btn = iot_button_create(&cfg);
    iot_button_register_cb(btn, BUTTON_LONG_PRESS_START, button_long_press_hold_cb, "hold");
}
```

```c
#define WAKEUP_PIN FEATURE_TOGGLE_PIN
#define WAKEUP_LEVEL 1

void init_deep_sleep(void)
{
    ESP_LOGI(TAG, "init deep sleep on pin %d", WAKEUP_PIN);
    esp_deep_sleep_enable_gpio_wakeup(WAKEUP_PIN, WAKEUP_LEVEL);
    printf("wakeup cause: %d\n", esp_sleep_get_wakeup_cause());
    switch (esp_sleep_get_wakeup_cause()) {
        case ESP_SLEEP_WAKEUP_GPIO: {
            printf("Wake up from GPIO\n");
            break;
        }
        default: {
            printf("Not a deep sleep reset\n");
        }
    }
}

void enter_deep_sleep(void)
{
    esp_deep_sleep_start();
}
```

chegewara, 2022-11-24:
You are using the same pin to wakeup and to start deep sleep. It never works.
You can try to test it this way:
```
void enter_deep_sleep(void)
{
vTaskDelay(2000); // 2s for example, to let time to release button and let the button debounce
    esp_deep_sleep_start();
}
```

robbietree8, 2022-11-25:
> You are using the same pin to wakeup and to start deep sleep. It never works. You can try to test it this way:
> 
> ```
> void enter_deep_sleep(void)
> {
> vTaskDelay(2000); // 2s for example, to let time to release button and let the button debounce
>     esp_deep_sleep_start();
> }
> ```

@chegewara thanks, your solution works well.

closed 2022-11-25 by robbietree8