espressif/esp-idf #10247: RTC_SW_SYS_RST when calling esp_restart on ESP32-C3

Kampi, 2022-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.2.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### What is the expected behavior?

Setting `SW_CPU_RESET` as `RESET_REASON`.

### What is the actual behavior?

The bootlog is reportin

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381f82
=> 0x40381f82: esp_restart_noos_dig at C:\Users\d.kampert_traplinked\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:44 (discriminator 1)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x71c
```

during each reboot.

### Steps to reproduce.

Code:

```
extern "C" void app_main(void)
{
    esp_log_level_set("gpio", ESP_LOG_NONE);
    ESP_LOGI("main", "Restarting...");
    esp_restart();
}
```

ESP-Marius, 2022-11-25:
Hi,

I think this behavior is as expected, if you check with `esp_reset_reason()` you should see that IDF indeed reports the reason as `ESP_RST_SW`.

The C3 ROM-code reset reasons doesnt actually have a `SW_CPU_RESET`, `esp_restart()` will have the ROM-code reset reason of either `RTC_SW_CPU_RESET` or `RTC_SW_SYS_RESET`, depending on if `CONFIG_ESP_SYSTEM_MEMPROT_FEATURE` is active or not.

Kampi, 2022-11-25:
Thanks. That was helpful and solved my issue.

closed 2022-11-25 by Kampi

namagregorio, 2025-07-06:
Hi, I have a similar problem:

19:11:53.793 -> ESP-ROM:esp32s3-20210327

19:11:53.793 -> Build:Mar 27 2021

19:11:53.793 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)

19:11:53.793 -> Saved PC:0x403cdad1

19:11:53.793 -> SPIWP:0xee

19:11:53.793 -> mode:DIO, clock div:1

19:11:53.793 -> load:0x3fce3818,len:0x508

19:11:53.793 -> load:0x403c9700,len:0x4

19:11:53.793 -> load:0x403c9704,len:0xad0

19:11:53.793 -> load:0x403cc700,len:0x29e4

19:11:53.793 -> entry 0x403c9880