espressif/esp-idf #10248: 对ESP32 ADC的几个疑问 以及 BUG反馈

kurau1, 2022-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

板子：ESP-WROOM-32
供电模式与烧录方式：USB
编译软件：VS CODE

本人因为项目需要，接触ESP32已经有2个月了，但是对ESP32的ADC仍然有一些疑问 以及 想反馈一些bug

### **问题一：**
ESP32真的能实现 真;2MSPS采样吗？

`IDFV4.4.2的dma_Read`、`F:\ESP-IDF\Espressif\frameworks\esp-idf-v4.4.2\components\driver\test\adc_dma_test下的test_esp32.c` 、`F:\ESP32\idf\esp-idf_5\esp-idf\components\driver\test_apps\i2s_test_apps\legacy_i2s_adc_dac`与 `IDFV5的continous_read_main` 四个工程我都尝试过了，我的结论是ESP32最多只能支持250KSPS采样。

我曾在乐鑫中文论坛找到过一篇关于实现2MSPS的实例，但其IDF已落后，多数API已无法在IDF4、IDF5中实现。链接如下:
https://www.esp32.com/viewtopic.php?f=25&t=8335&sid=ef3ed33a1d70f59626992ff3cb8dcc52

说说我为什么得出ESP32只支持250KSPS的结论。
四个工程，在程序的正确性上来说都能将采样率提高到2MSPS，但实际采样得到的数据却存在重复出现的情况。
例如我将采样率设置为1MSPS，通过测试，采样率的确在1MSPS附近，也能得到近1M个数据，但这些数据中，会有相邻四个数据相同，这不就意味着我得到的数据真实性必须除以4吗？1MSPS/4 = 250KSPS。以下有2个IDF版本得到的相同结果：
以下是IDF4.4.2
[image]

以下是IDF5
[image]

2者都是1MSPS下采样的结果，可见还有一个问题存在:IDF5的实测采样率 比IDF4.4.2略低，只有800多K。
若我将采样率设置为250KSPS，重复数据的情况就会消除：
[image]

且多开通道重复出现的情况也不会消除，并不是跟我曾经想的一样，八个通道分别250KSPS，所以加起来2MSPS.在我看来，ESP32采样I2S去采样ADC，但ADC采样率并没有这么快（在250KSPS附件），导致ESP32多次获取到相同的值（因为ADC未来得及转换）

### **问题二：**
ESP32单通道采样时，IDF4.4.2中ad采样数据前后顺序，这个问题在我前面发的中文论坛链接里也有提到，原话：`我怀疑单通道AD采样值被i2s模块切成左右声道，输出时前后顺序错位，i2s改为单声道可能就OK了。`

这是之前我使用ESP32采样的正弦波波形图，我一度怀疑这是ESP32 ADC内部噪声所致。
[image]

根据论坛中的建议，我调换了一些数据的前后顺序，下方是更改后的波形图：
[image]
#10058这里也有提到

IDF5我还未进行测试，不知是否改善。

### **结论**
我已经在ESP32的ADC采样率上花费了太多时间，但始终无法在ESP32内部ADC高速采样上获得进展，因此发布了这篇帖子。希望碰到过类似问题的朋友给些建议。
如何才能实现ESP32的2MSPS采样。

kurau1, 2022-11-26:
@Icarus113  Hello, I often see you answering questions about ADCs, if you have time, can you take a look at my questions? I've been bothered by this issue for a long time.
Thanks

你好，我经常看到你在回答关于ADC的问题，如果有时间的话，能看看我的问题吗？我已经被这个问题困扰很久了。
谢谢你

FlynnMa, 2022-11-27:
这是一个很有质量的问题，看得出来你做了很多研究，给你点个赞，

但是我怀疑有人能回答你的问题，从乐鑫的test cases可以看出来，他们自己没有做过你所做过的相关的测试。我建议你自己根据user mannul里面列出来的寄存器再做一些尝试，但是他们的文档不全，引用网友的话：
"esp32 is chinese chip, which is not well documented.This product looks promising online, but the documentation is not useful."

如果你已经花了很多时间研读他们的user manual，还是无法解决，换其他主芯片吧。

kurau1, 2022-11-27:
> This is a great question with a lot of investigations, I am using esp32c3, I got same problem on this

C3不会吧，ESP32系列除了ESP32是用I2S采样ADC理论能达到2MSPS，其他的都只有83333Hz，没有这个重复的问题。
你有其他的ADC问题可以问我，我知道的话会回答你，哈哈哈。（你就用中文吧，我看你一直在编辑你这条消息）。

Icarus113, 2022-11-27:
@kutau1 sorry 给你使用带来困扰。。这个问题确实很奇怪，我们正在查。有更新了我会在这里同步的。

@FlynnMa this is a known issue on c3. The frequency is limited, please check [this](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/soc/esp32c3/include/soc/soc_caps.h#L96)

kurau1, 2022-11-27:
> @kutau1 sorry 给你使用带来困扰。。这个问题确实很奇怪，我们正在查。有更新了我会在这里同步的。
> 
> @FlynnMa this is a known issue on c3. The frequency is limited, please check [this](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/soc/esp32c3/include/soc/soc_caps.h#L96)

真的麻烦你了，帮我看看吧，这个bug确实是存在的。

Thanks！

FlynnMa, 2022-11-28:
> river\test\a

As github is a global platform, using Chinese is not frendly, I will read @kurau1 posted link, if I have further problem, I will direct message you, thanks.

Icarus113, 2022-11-28:
@FlynnMa 
Ops, sorry pasted wrong link there. This [SOC_ADC_SAMPLE_FREQ_THRES_HIGH](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32c3/include/soc/soc_caps.h#L96) is what I posted. (Have modified the link in my previous post as well)

Icarus113, 2022-11-28:
@kurau1  
Regarding to your question 1: 
Yes, your observation is right. Current [max freq limitation](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/soc_caps.h#L122) on ESP32 is wrong. 

It's a hardware limitation, and similar to the condition on other ESP chips. You can see on other chips, the high limitation is 83333 Hz. The limitation comes from the SAR ADC(s). The behaviours are similar among chips, if you use a frequency higher than the high limitation, you'll get repeated conversion results.

We will update this macro on ESP32 later, but it takes some time. I don't know the correct high limitation on ESP32 now, either. What I can suggest now is using it under lower frequency, possibly you can refer to the 83333 Hz.

kurau1, 2022-11-28:
@Icarus113 
Thanks for your reply.
From my experience.  SOC_ADC_SAMPLE_FREQ_THRES_HIGH of ESP32 maybe is around 250KHz, 
Below is my test picture. When i set 270Khz and open 6 channel、7 channel. there are no repeated resluts
[image]

When i set 280Khz and open 6 channel、7 channel. there are  repeated resluts.
[image]

Another reason is when i set frequency is 1M, i will get 4 repeated results. when i set frequency is 2M, i will get 8 repeated results. 
Now, i know it's a hardware limitation. So i won't spend my time on software. 

At last, Thanks for your experiment and reply.

hongwuea, 2023-08-27:
dalao这两个问题，我在idf的5.1版本上都很好的复现了，xs，完全没有解决。
我从idf5.1的连续读取例程出发，改出了一个简单的连续读取后串口打印的小程序，然后在电脑端用python接收
[image]
第一张图用2Mbps读取一个通道的20kHz正弦，重现了问题1，也不知道是它ADC采样保持就固定250kbps还是能搁别的地方配置。
[image]
第二张图用200kbps读取一个通道的10kHz正弦，显示了读取的交错现象，这个交错现象在第一张图的台阶里其实也能观测到。
[image]

ELFsay, 2025-08-02:
This is a problem with the default frequency division coefficient of esp-idf（The version I use is V5.1.4）. Can this code be used to solve it, but the cost may be reduced accuracy? I am not sure. I am currently using this code to solve the motor current sampling problem. I have reported this to the engineer in the group before, but he did not respond to me.
`    if (sample_freq_hz >= 1000000)
        // {
        // adc_ll_set_sample_cycle(2);
        adc_ll_digi_set_clk_div(2);
    // }
    else if (sample_freq_hz > 500000)
        adc_ll_digi_set_clk_div(4);
    else
        adc_ll_digi_set_clk_div(8);`