espressif/esp-idf #10249: Why PPP is new/experimental?

WangShuoran, 2022-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, Dear espressif developer:

I am lwip runoob. Excuse me.
When I want to use ESP32S3 connect CAT1(EC600N) with usb, but I find CONFIG_LWIP_PPP_SUPPORT is new/experimental, I don't know why. 
Does LwIP's PPP is not stable? Or only espressif not test this function?
And if I want to let this function more stable, what I can do?

Thank you.

WangShuoran, 2022-11-25:
Does esp-idf use lwip version is 2.1.2(When I find components/lwip/lwip/CHANGLOG is 2.1.2)? 
And I find the PPP is in addons https://www.nongnu.org/lwip/2_1_x/group__ppp.html. So I don't know PPP is new/experimental why?

david-cermak, 2023-01-04:
Hi @WangShuoran 

Sorry for the late reply. The `PPP` is officially supported by Espressif, sorry for the confusion. We forgot to remove this note about the *experimental* status.
Support for GSM modem is provided by [esp-modem](https://components.espressif.com/component/espressif/esp_modem) component, but you can also use the PPP(oS) directly from lwIP, or via `esp-netif` abstraction here in ESP-IDF:

https://github.com/espressif/esp-idf/blob/4c98bee8a4417d92a387ef2416be14ff1fdfa92a/components/esp_netif/include/esp_netif_defaults.h#L132-L140

WangShuoran, 2023-01-05:
Thank you, **So, at IDFv5.1 or other version will remove the experimental status**? I don't know am  I exactly use [esp-modem](https://components.espressif.com/component/espressif/esp_modem). Can you help me with the following contents?
Because of when I test IoT-solution's USB 4G model with EC600N, I find any status in ESP32 open can not connect with EC600N(I do not know if this is the case, but at least I cann't use EC600N outbound). 
I don't know the [IoT-solution's USB 4G model](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module) and [ESP-modem](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem) different or relation. You know https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples is very complex with runoob, and you give me last example:
[image]
I don't know want I should see at first. Does ESP-MODEM relation example is use serial port?

closed 2023-01-13 by espressif-bot