espressif/esp-idf #10252: scan_evt timeout BLE_HS_FOREVER

jsw-davidhuang, 2022-11-26:
i start a forever scan as a ble host using NIMBLE
`
  disc_params.filter_duplicates = 0;

    /**
     * Perform a passive scan.  I.e., don't send follow-up scan requests to
     * each advertiser.
     */
    disc_params.passive = 0;

    /* Use defaults for the rest of the parameters. */
    disc_params.itvl = 0;
    disc_params.window = 0;
    disc_params.filter_policy = 0;
    disc_params.limited = 0;
rc = ble_gap_disc(own_addr_type, BLE_HS_FOREVER, &disc_params,
                      blecent_gap_event, NULL);
`

when receive mqtt cmd to connect device, first stop scan
`
rc = ble_gap_disc_cancel();
    if (rc != 0) {
        ESP_LOGI(TAG, "Failed to cancel scan; rc=%d\n", rc);
        return -1;
    }
`

when finish, i will restart scan just like the first step

but when i run the device for 2 days later, i try to connect the ble device again, i got the error

`
if (rc != 0) {
        ESP_LOGI(TAG, "Error: Failed to connect to device; "addr=%s; rc=%d\n", addr_str(lock_addr.val), rc);
        return -3;
    }
`

the error code is 19

[image]

how can i fix this issue?

jsw-davidhuang, 2022-11-28:
SDK version:v4.3-194-g5b5e46971-dirty

rahult-github, 2022-11-28:
Hi @jsw-davidhuang ,

Can you please share some more information, which can assist us to debug this further.  The error code 19 implies a HCI command timed out. But the current set of logs don't give much information about what has gone wrong. Can you possibly help share:

1. Which chipset you are using ?
2. As mentioned "but when i run the device for 2 days later"  , does this imply there was no other activity apart from discovery ?  Were there any devices in the vicinity with its advertising enabled that were being detected in the discovery activity ? 
3. Was connection the only operation done OR any other operation was tried too ?
4. Did the next operation of starting discovery work ? 
5. Is it possible to enable debug logs at your end and share , if feasible?

jsw-davidhuang, 2022-11-28:
Hi thankyour for your reply,
1. i am using ESP32-WROOM-32E
2. yes, just let the device keep discovering ;  yes, the usage scenario would be like this:
2.1: i use esp32-ble-host to continus discover the adversitement package from our ble lock device which will continuely advertise the lock state value in the adv package. so, when the lock is opened, esp32 can know from the advertisment data without connecting to the lock, then report to cloud using mqtt. 
2.2: i can also send command from cloud to esp32 to connect to locks by ble, then bypass the command from cloud to device.

3. it is the only operation. when i found this issue, i just use the 2.2 step to start connect to device. this is where i found the issue, cause i found esp32 always return false to my connect command.
4. no, when i restart the scan, first it did not report error(i think it is async), then i got the print: scan_evt timeout
5. the picture is the only error message i got from log, all other logs seems normal.

i do a lot of research too. please see if it is helpful.
1. it happen when using wifi and ble at the same time?
2. it there any way to just reset ble-host function, so that i can do reset ble when this error happen. i try to use this function and it not helpful:
`
ble_hs_sched_reset(19);
`
4. is there a ble watchdog function?

SumeetSingh19, 2022-11-29:
Hi @jsw-davidhuang ,
1. I cannot tell this based on current information
2. Yes, you can stop deinit nimble and start again. For reference, you can refer to https://github.com/espressif/esp-idf/blob/release/v4.4/examples/bluetooth/nimble/blecent/main/main.c#L530 how to deinit / init stack
3. Stack has asserts , whereas controller code has watchdog. In this case, it is not stuck ( i.e. it is not gone in a continous loop and thereby hogging the system) .So this case will not cause any watchdog.

Having said , since you are getting console logs, we request you to re-build application with debug log enabled like below  and share log

1. idf,py menuconfig > Component config > Bluetooth > NimBLE options > NimBLE Host log verbosity > Set it to "Debug logs"
5. idf.py menuconfig > Component config > Log Output > Default log verbosity > Set it to "Debug"

Save and quit the menuconfig, build, flash, and monitor and you should see log output starting with D
This means that debug logs are enabled. You may need to re-enable them when building the application again.

Share the entire log if you can and not just where the error occurs please.

jsw-davidhuang, 2022-12-01:
[fail_log.txt](https://github.com/espressif/esp-idf/files/10128199/fail_log.txt)
hi @SumeetSingh19 , thanks , i do as what you said, and here is the log file, please see and check if anything wrong

SumeetSingh19, 2022-12-05:
Hi @jsw-davidhuang,
I checked the logs, and it's difficult to pinpoint the problem without reproducing the issue on my end. Could you please provide the application you are using on ESP32.
Waiting for two days is also a long time, have you tried waiting for just 5-15 mins and checking if the problem still exists.

jsw-davidhuang, 2022-12-06:
> Hi @jsw-davidhuang, I checked the logs, and it's difficult to pinpoint the problem without reproducing the issue on my end. Could you please provide the application you are using on ESP32. Waiting [...]

[build.zip](https://github.com/espressif/esp-idf/files/10159274/build.zip)

i am not sure which file you need, i copy as many as much in the build folder, in my side i had test 26h, and this issue come out.

after flash, it will going to ble host scan mode, and keep printing the advisterment data it can search around. in my side, i found no print anymore after 26 hours

SumeetSingh19, 2022-12-09:
Hi @jsw-davidhuang,
I found a similar test case in our internal testing that does discovery for a long time and then stops, but there's no attempt at connection in the test case and the discovery succeeds.

Also the files you provided are simply the build folder inside the application, could you please send the entire application. What I mean is the folder that contains the build folder and the main folder. Thanks.

jsw-davidhuang, 2022-12-10:
> 

of course, do you got your email so that i can send to you?

SumeetSingh19, 2022-12-12:
@jsw-davidhuang 

Can you zip your application and send it, or if the project is a github repo you can send me the link, if either of these options don't work for you, I can share my email with you.

jsw-davidhuang, 2022-12-12:
> 

@SumeetSingh19 , i upload to google drive,  https://drive.google.com/file/d/1VayStjGjpVrWw8nGdJLvMhP8GMX7cswY/view?usp=share_link,  please download from here

jsw-davidhuang, 2022-12-14:
i also test another way, i start a 10 second scan, then got a discover complete then immedially start scan again. i found that when it run for a while it cannot got the discover complete event again. like this:

[image]
[image]

jsw-davidhuang, 2022-12-26:
> @jsw-davidhuang
> 
> Can you zip your application and send it, or if the project is a github repo you can send me the link, if either of these options don't work for you, I can share my email with you.

hi @SumeetSingh19 any update on this? i had test the same code in esp-idf 5.0 now, and the problem did not occure anymore at least for 3.6 days. but i hope you guys can still fix it in the esp-idf 4.3

thorrak, 2023-03-06:
Is it still stable, @jsw-davidhuang ?

I’m definitely encountering this in the v4.4.4-based Arduino framework for ESP32.

jsw-davidhuang, 2023-03-07:
> Is it still stable in v5.x, @jsw-davidhuang ?
> 
> I’m definitely encountering this in the v4.4.4-based Arduino framework for ESP32.

yes, a few month pass now, it is stable

closed 2023-03-07 by jsw-davidhuang