espressif/esp-idf #10254: ESP32 reset after call LVGL delete function inside Nimble event

jiabuda, 2022-11-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF: 4.4.2
LVGL: 8.3.3
Chip: ESP32

Hi, I am integrating Nimble and LVGL together to build a IOT device. Screen would be updated after Nimble central role discovery a peripheral device and establish connection. So inside Nimble event `BLE_GAP_EVENT_CONNECT` , I call some lvgl function to switch to another screen and call `lv_obj_del`to delete the current screen. But the mcu reset and give a confusing core dump.

Here is the code:
```
case BLE_GAP_EVENT_CONNECT:
        /* A new connection was established or a connection attempt failed. */
        if (event->connect.status == 0)
        {
            /* Connection successfully established. */
            ESP_LOGI(TAG, "Connection established");

            rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
            assert(rc == 0);

            /* Remember peer. */
            rc = peer_add(event->connect.conn_handle);
            if (rc != 0)
            {
                ESP_LOGE(TAG, "Failed to add peer; rc=%d", rc);
                return 0;
            }

            /* Perform service discovery. */
            rc = peer_disc_all(event->connect.conn_handle, blecent_on_disc_complete, NULL);
            if (rc != 0)
            {
                ESP_LOGE(TAG, "Failed to discover services; rc=%d", rc);
                return 0;
            }          
        }
        else
        {
            /* Connection attempt failed; resume scanning. */
            ESP_LOGE(TAG, "Error: Connection failed; status=%d", event->connect.status);
        }
        return 0;
```

and 
```
static void blecent_on_disc_complete(const struct peer *peer, int status, void *arg)
{

    if (status != 0)
    {
        /* Service discovery failed.  Terminate the connection. */
        ESP_LOGE(TAG, "Error: Service discovery failed; status=%d conn_handle=%d", status, peer->conn_handle);
        ble_gap_terminate(peer->conn_handle, BLE_ERR_REM_USER_CONN_TERM);
        return;
    }

    /* Service discovery has completed successfully.  Now we have a complete
     * list of services, characteristics, and descriptors that the peer
     * supports.
     */
    ESP_LOGI(TAG, "Service discovery complete; status=%d conn_handle=%d", status, peer->conn_handle);

    /* Now perform three GATT procedures against the peer: read,
[... 110 lines cut ...]
0x400d6cad: peer_dsc_disced at D:/git/giiker/esp32_nimble/components/ble_gatt/peer.c:235

0x400d8987: ble_gattc_disc_all_dscs_cb at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:2577

0x400d89bf: ble_gattc_disc_all_dscs_err at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:2646

0x400d9cf5: ble_gattc_rx_err at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4387

0x400da2f5: ble_att_clt_rx_error at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:46

0x400d858d: ble_att_rx at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x400d844e: ble_hs_hci_evt_acl_process at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:927

0x400d7b45: ble_hs_process_rx_data_queue at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x400d7b57: ble_hs_event_rx_data at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

0x400e06da: ble_npl_event_run at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x400d63a6: blecent_host_task at D:/git/giiker/esp32_nimble/components/ble_gatt/central.c:377

0x40093cc1: vPortTaskWrapper at C:/Users/Endy/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

chegewara, 2022-11-26:
Currently I am playing with lvgl and nimble on arduino and S3. With simple screens I have no problem switching between screens.
Another application has more extended screens and transition caused crash. Unlucky to me no UART, so no backtrace.
It turned out that simple task to handle lvgl was the problem. Running it on core 1 crashed app, now its running on core 0 and all is fine.

jiabuda, 2022-11-27:
> Currently I am playing with lvgl and nimble on arduino and S3. With simple screens I have no problem switching between screens. Another application has more extended screens and transition caused [...]

Hi chegewara, i ask this question in lvgl, they reply that need to take the semaphore when you update the screen then give it again. i try this and problem solved.

closed 2022-11-27 by jiabuda